espressif/esp-idf #13111: ESP_MESH Using excesive heap memory

luisyague, 2024-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi everyone, I am using an ESP32 with esp-idf v4.3.7. We are planning to deploy some esp32s with wifi-mesh functionality, using esp-mdf on top of it. However we are seeing that the esp-mesh uses a huge amount of heap memory (around 60ks). Iis the any way to reduce this heap usage?  
Thanks in advance.

zhangyanjiaoesp, 2024-02-06:
@luisyague 
 The esp-idf v4.3.7 is already end of life, suggest you use release/v4.4 or later branch. And the esp-mdf is limited maintain, recommend to use https://github.com/espressif/esp-mesh-lite or just base on the mesh example in esp-idf. Currently, there is no way to reduce the memory usage of wifi mesh.

MaxwellAlan, 2024-02-20:
Hi @luisyague 

We will close this issue since the release/v4.3 is EOL.

Thanks.

closed 2024-02-20 by MaxwellAlan