espressif/esp-idf #13112: c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.peripherals.ld: No such file or directory

Yan19960220, 2024-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Windows

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

idf.py --no-ccache build 
build sucess and generate 3 bin files

### What is the actual behavior?

C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2>idf.py --no-ccache build
Executing action: all (aliases: build)
Running ninja in directory C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D C:\Users\Yan\Desktop\esp\tena_health-ten...na_health-tena-firmware-tena_v2/build/bootloader/bootloader.bin
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[5/7] Linking CXX executable tena.elfFAILED: tena.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/Users/Yan/Desktop/esp/tena_health-tena-firmware-tena_v2/build/tena.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\tena.elf.rsp -o tena.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.peripherals.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build\log\idf_py_stderr_output_8468 and C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build\log\idf_py_stdout_output_8468

### Steps to reproduce.

follow the guidance:

idf.py build

### Build or installation Logs.

```plain
C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2>idf.py --no-ccache build
Executing action: all (aliases: build)
Running ninja in directory C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build
Executing "ninja all"...
[1/1] cmd.exe /C "cd /D C:\Users\Yan\Desktop\esp\tena_health-ten...na_health-tena-firmware-tena_v2/build/bootloader/bootloader.bin
Bootloader binary size 0x6810 bytes. 0x7f0 bytes (7%) free.
[5/7] Linking CXX executable tena.elfFAILED: tena.elf
cmd.exe /C "cd . && C:\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-g++.exe -mlongcalls -Wno-frame-address -Wl,--cref -Wl,--defsym=IDF_TARGET_ESP32=0 -Wl,--Map=C:/Users/Yan/Desktop/esp/tena_health-tena-firmware-tena_v2/build/tena.map -Wl,--no-warn-rwx-segments -fno-rtti -fno-lto -Wl,--gc-sections -Wl,--warn-common -T esp32.peripherals.ld -T esp32.rom.ld -T esp32.rom.api.ld -T esp32.rom.libgcc.ld -T esp32.rom.newlib-data.ld -T esp32.rom.syscalls.ld -T esp32.rom.newlib-funcs.ld -T memory.ld -T sections.ld @CMakeFiles\tena.elf.rsp -o tena.elf  && cd ."
c:/espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/12.2.0/../../../../xtensa-esp32-elf/bin/ld.exe: cannot open linker script file esp32.peripherals.ld: No such file or directory
collect2.exe: error: ld returned 1 exit status
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build\log\idf_py_stderr_output_8468 and C:\Users\Yan\Desktop\esp\tena_health-tena-firmware-tena_v2\build\log\idf_py_stdout_output_8468
```

Yan19960220, 2024-02-05:
I have used "git submodule --init --recursive" to generate esp32.peripherals.ld in soc/ld.

o-marshmallow, 2024-02-06:
Hello @iXiaoxideYan ,

How did you download and install ESP-IDF? Which procedure did you follow?

The `esp32.peripherals.ld` file is not part of a submodule, it should be pulled automatically in `components/soc/esp32/ld/` when you `git clone` the ESP-IDF repository.

iXiaoxideYan, 2024-02-06:
> Hello @iXiaoxideYan ,
> 
> How did you download and install ESP-IDF? Which procedure did you follow?
> 
> The `esp32.peripherals.ld` file is not part of a submodule, it should be pulled automatically in `components/soc/esp32/ld/` when you `git clone` the ESP-IDF repository.

> Hello @iXiaoxideYan ,
> 
> How did you download and install ESP-IDF? Which procedure did you follow?
> 
> The `esp32.peripherals.ld` file is not part of a submodule, it should be pulled automatically in `components/soc/esp32/ld/` when you `git clone` the ESP-IDF repository.

Just to let you know that I have replied to your comment.

jakub-kocka, 2024-02-06:
Hello @iXiaoxideYan,
can you please send the output from `git status`?

If all components are updated thanks to `git submodule update`, please try if the `.ps1` scripts will fix the issue (and if there is no special need to use "install.bat \<target\>" syntax you can use just install and export).

    .\install.ps1
    . .\export.ps1

dobairoland, 2024-02-06:
Can you please try to build an example project included with ESP-IDF? Do you see the same issue? If yes then please share the full output log.

iXiaoxideYan, 2024-02-13:
Done. It works!

> Hello @iXiaoxideYan, can you please send the output from `git status`?
> 
> If all components are updated thanks to `git submodule update`, please try if the `.ps1` scripts will fix the issue (and if there is no special need to use "install.bat <target>" syntax you can use just install and export).
> 
> ```
> .\install.ps1
> . .\export.ps1
> ```
> 
> from Windows PowerShell.

o-marshmallow, 2024-02-13:
@iXiaoxideYan Thanks for the confirmation, closing the issue.

closed 2024-02-13 by o-marshmallow