espressif/esp-idf #13113: libcoexist functions missing

KaeLL, 2024-02-05:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Building my app on `v5.1.2-684-g7cbee80fb9`, I get the error
```
/esp-idf/components/esp_coex/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.17+0x0): undefined reference to `btdm_rf_bb_reg_init'
/esp-idf/components/esp_coex/lib/esp32/libcoexist.a(coexist_hw.o):(.iram1.17+0x16): undefined reference to `btdm_rf_bb_reg_init'
```
<details>
<summary>defconfig</summary>

```
CONFIG_BOOTLOADER_COMPILER_OPTIMIZATION_PERF=y
CONFIG_BOOTLOADER_LOG_LEVEL_NONE=y
CONFIG_BOOTLOADER_APP_ROLLBACK_ENABLE=y
CONFIG_BOOTLOADER_SKIP_VALIDATE_ALWAYS=y
CONFIG_ESPTOOLPY_FLASHMODE_QIO=y
CONFIG_ESPTOOLPY_FLASHFREQ_80M=y
CONFIG_ESPTOOLPY_FLASHSIZE_4MB=y
CONFIG_ESPTOOLPY_AFTER_NORESET=y
CONFIG_PARTITION_TABLE_CUSTOM=y
CONFIG_PARTITION_TABLE_CUSTOM_FILENAME="partitions_ota.csv"
CONFIG_PARTITION_TABLE_OFFSET=0xC000
CONFIG_COMPILER_WARN_WRITE_STRINGS=y
CONFIG_BT_ENABLED=y
CONFIG_BT_BTU_TASK_STACK_SIZE=4096
CONFIG_BT_BLE_SMP_ENABLE=n
CONFIG_BT_STACK_NO_LOG=y
CONFIG_BT_ACL_CONNECTIONS=1
CONFIG_BT_MULTI_CONNECTION_ENBALE=n
CONFIG_BT_BLE_DYNAMIC_ENV_MEMORY=y
CONFIG_BT_BLE_HOST_QUEUE_CONG_CHECK=y
CONFIG_BTDM_CTRL_BLE_MAX_CONN=1
CONFIG_BTDM_CTRL_MODEM_SLEEP=n
CONFIG_BTDM_BLE_SCAN_DUPL=n
CONFIG_BTDM_BLE_ADV_REPORT_FLOW_CTRL_SUPP=n
CONFIG_SPI_MASTER_IN_IRAM=y
CONFIG_SPI_SLAVE_ISR_IN_IRAM=n
CONFIG_TWAI_ERRATA_FIX_BUS_OFF_REC=n
CONFIG_TWAI_ERRATA_FIX_TX_INTR_LOST=n
CONFIG_TWAI_ERRATA_FIX_RX_FRAME_INVALID=n
CONFIG_TWAI_ERRATA_FIX_RX_FIFO_CORRUPT=n
CONFIG_TWAI_ERRATA_FIX_LISTEN_ONLY_DOM=n
CONFIG_GPTIMER_SUPPRESS_DEPRECATE_WARN=y
CONFIG_ESP_TLS_INSECURE=y
CONFIG_ESP_TLS_SKIP_SERVER_CERT_VERIFY=y
CONFIG_ETH_DMA_BUFFER_SIZE=256
CONFIG_ETH_DMA_RX_BUFFER_NUM=3
CONFIG_ETH_DMA_TX_BUFFER_NUM=3
CONFIG_ETH_USE_SPI_ETHERNET=n
CONFIG_ESP_EVENT_POST_FROM_ISR=n
CONFIG_ESP_HTTP_CLIENT_ENABLE_BASIC_AUTH=y
CONFIG_ESP_HTTPS_OTA_ALLOW_HTTP=y
CONFIG_ESP32_REV_MIN_1=y
CONFIG_RTC_CLK_SRC_INT_8MD256=y
CONFIG_ESP_PHY_CALIBRATION_AND_DATA_STORAGE=n
CONFIG_ESP_PHY_REDUCE_TX_POWER=y
CONFIG_ESP_DEFAULT_CPU_FREQ_MHZ_240=y
CONFIG_ESP_SYSTEM_PANIC_PRINT_HALT=y
CONFIG_ESP_MAIN_TASK_STACK_SIZE=10240
CONFIG_ESP_CONSOLE_UART_CUSTOM=y
CONFIG_ESP_CONSOLE_UART_BAUDRATE=921600
CONFIG_ESP_INT_WDT_TIMEOUT_MS=30
CONFIG_ESP_TASK_WDT_PANIC=y
CONFIG_ESP_TASK_WDT_TIMEOUT_S=10
CONFIG_ESP_DEBUG_OCDAWARE=n
CONFIG_ESP_IPC_TASK_STACK_SIZE=1536
CONFIG_ESP_WIFI_NVS_ENABLED=n
CONFIG_ESP_WIFI_IRAM_OPT=n
CONFIG_ESP_WIFI_RX_IRAM_OPT=n
CONFIG_FREERTOS_HZ=200
CONFIG_FREERTOS_TASK_FUNCTION_WRAPPER=n
CONFIG_FREERTOS_WATCHPOINT_END_OF_STACK=y
CONFIG_FREERTOS_ISR_STACKSIZE=2096
CONFIG_FREERTOS_CORETIMER_1=y
CONFIG_HEAP_ABORT_WHEN_ALLOCATION_FAILS=y
CONFIG_LOG_DEFAULT_LEVEL_DEBUG=y
CONFIG_LWIP_TCP_SND_BUF_DEFAULT=5744
CONFIG_LWIP_TCP_WND_DEFAULT=5744
CONFIG_LWIP_SNTP_MAX_SERVERS=4
CONFIG_MBEDTLS_CERTIFICATE_BUNDLE_DEFAULT_CMN=y
CONFIG_MBEDTLS_TLS_CLIENT_ONLY=y
CONFIG_MQTT_TRANSPORT_WEBSOCKET=n
CONFIG_NVS_ASSERT_ERROR_CHECK=y
CONFIG_SPI_FLASH_ENABLE_ENCRYPTED_READ_WRITE=n
CONFIG_ULP_COPROC_ENABLED=y
CONFIG_ULP_COPROC_RESERVE_MEM=128
CONFIG_WIFI_PROV_BLE_BONDING=y
CONFIG_WIFI_PROV_BLE_FORCE_ENCRYPTION=y
CONFIG_EMAC_RECV_TASK_CORE=0
```

</summary>

esp-lrh, 2024-02-06:
Hi, @KaeLL 
Please run `git submodule update` and `git status` in IDF directory, make sure that there is no change in the subrepository `components/bt/controller/lib_esp32`.

Run `git log` in directory `components/bt/controller/lib_esp32`, the first commit should be `b0f132f1`.

KaeLL, 2024-02-06:
Just updated the branch, btw.
```
$ git describe --tags && git status && git submodule status
v5.1.2-691-g7380f96017
On branch release/v5.1
Your branch is up to date with 'origin/release/v5.1'.

nothing to commit, working tree clean
 24c60e243580c7868f4334a1ba3123481fe1aa48 components/bootloader/subproject/components/micro-ecc/micro-ecc (v1.1)
 b0f132f17ed48553a4638344023d44af61b7c20a components/bt/controller/lib_esp32 (remotes/origin/HEAD)
 7f3eb5f5568439cdb32dfe9dbf69d99e9c72e69f components/bt/controller/lib_esp32c2/esp32c2-bt-lib (remotes/origin/HEAD)
 9d40b3bc0ac38d72b66fe0f612a8dda64c7330d7 components/bt/controller/lib_esp32c3_family (heads/master-21-g9d40b3b)
 14bb72fed1ec294cf9373005c13d2d9979c72a09 components/bt/controller/lib_esp32c6/esp32c6-bt-lib (heads/main-45-g14bb72f)
 ec71faf431106c29c05f553c7c8719542208c561 components/bt/controller/lib_esp32h2/esp32h2-bt-lib (remotes/origin/HEAD)
 31335a5a4fd0e063edc086d3c268399404db3bfd components/bt/host/nimble/nimble (remotes/origin/master-1547-g31335a5a)
 eeecc49ce8af123cf8ad40efdb9673e37b56230f components/cmock/CMock (v2.5.2-2-geeecc49)
 496ee81e168881cad93f9a813802e7f7a6daec92 components/esp_coex/lib (remotes/origin/HEAD)
 2d319a382336cf0522ea4bb5a3fbd6701a8633c6 components/esp_phy/lib (remotes/origin/HEAD)
 0a25c10a587df080550995e18fc10c996abc7ecf components/esp_wifi/lib (remotes/origin/release/v5.1)
 d2e28f872472ffc6a704faae65ddee1f24e2dfba components/heap/tlsf (heads/idf-2-gd2e28f8)
 87d8f0961a01bf09bef98ff89bae9fdec42181ee components/json/cJSON (v1.7.15-21-g87d8f09)
 542ba2997f6cc14fe9c3d781bf9b0d890cd70bb1 components/lwip/lwip (remotes/origin/master-281-g542ba299)
 dd91f30058dfad68d80267cc82cd8472c4aa54b5 components/mbedtls/mbedtls (mbedtls-3.5.0-5-gdd91f3005)
 e6afdb4025fe018ae0add44e3c45249ea1974774 components/mqtt/esp-mqtt (ESP-MQTT_FOR_IDF_4.x-227-ge6afdb4)
 fcee30db4b7342de4df9105bdd049a09d2d63187 components/openthread/lib (heads/master-21-gfcee30d)
 456c448284486abe2a9118a9fdaa468fe2383fcd components/openthread/openthread (thread-reference-20230706-340-g456c44828)
 abc67a11c6db271bedbb9f58be85d6f4e2ea8389 components/protobuf-c/protobuf-c (v1.4.1)
 0dbb3f71c5f6fae3747a9d935372773762baf852 components/spiffs/spiffs (0.2-255-g0dbb3f7)
 7d2bf62b7e6afaf38153041a9d53c21aeeca9a25 components/unity/unity (v2.4.3-51-g7d2bf62)
```

esp-lrh, 2024-02-07:
Run `idf.py fullclean`  and build your app again, paste the full log here. 

In addition, please try to build `examples/bluetooth/nimble/bleprph_wifi_coex` for esp32, if build failed, paste the full log here too.

The two missing functions were added in commit `b0f132f1` of `components/bt/controller/lib_esp32`, I remove that commit and build the coexist example, got the same error with you.

If your app build failed and the `examples/bluetooth/nimble/bleprph_wifi_coex` build success, checkout idf to `db1e54a0`, run `git submodule update` and build your app again,  this version of idf will not include any reference to the two missing functions.

KaeLL, 2024-02-07:
> Run idf.py fullclean and build your app again, paste the full log here.
<details>
<summary>$ idf.py fullclean all</summary>

```
$ idf.py fullclean all
Executing action: fullclean
Executing action: remove_managed_components
Executing action: all (aliases: build)
Running cmake in directory /project/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/tools/python_env/idf5.1_py3.11_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=1 /project"...
-- IDF_TARGET is not set, guessed 'esp32' from sdkconfig '/project/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.40.1") 
-- Component directory /project/components/config_ar does not contain a CMakeLists.txt file. No component will be added
-- ccache will be used for faster recompilation
-- The C compiler identification is GNU 12.2.0
-- The CXX compiler identification is GNU 12.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /tools/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /project/sdkconfig
/project/sdkconfig:749 CONFIG_ESP_SLEEP_DEEP_SLEEP_WAKEUP_DELAY was replaced with CONFIG_ESP_SLEEP_WAIT_FLASH_READY_EXTRA_DELAY
Compiler supported targets: xtensa-esp32-elf

-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Found Python3: /tools/python_env/idf5.1_py3.11_env/bin/python (found version "3.11.6") found components: Interpreter 
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS
-- Performing Test C_COMPILER_SUPPORTS_WFORMAT_SIGNEDNESS - Success
-- App "ACCESSRUN_ATS_FIRMWARE" version: 1.1.0
-- Adding linker script /project/build/esp-idf/esp_system/ld/memory.ld
-- Adding linker script /esp-idf/components/esp_system/ld/esp32/sections.ld.in
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-data.ld
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.syscalls.ld
-- Adding linker script /esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Adding linker script /project/build/esp-idf/ar_ulp_reset/ulp_ar_ulp_reset/ulp_ar_ulp_reset.ld
-- Components: access_request app_trace app_update ar-access-req ar-cfg-page ar-config ar-log ar-mqtt ar-wifi ar_relay ar_ulp_reset bootloader bootloader_support bsp bt cmock console crypt_ar cxx driver efuse emergency_button error_check_ar esp-tls esp_adc esp_app_format esp_coex esp_common esp_eth esp_event esp_gdbstub esp_hid esp_http_client esp_http_client_ar esp_http_server esp_https_ota esp_https_server esp_hw_support esp_lcd esp_local_ctrl esp_mm esp_netif esp_netif_stack esp_partition esp_phy esp_pm esp_psram esp_ringbuf esp_rom esp_system esp_timer esp_wifi espcoredump esptool_py fatfs freertos gpi_ar gpio_ext gpio_user hal heap http_ar http_parser http_parser_ar i2c id_bio idf_test ieee802154 json json_utils log lwip main mbedtls memory_tracer mqtt network newlib nfc nvs_flash openthread ota partition_table perfmon protobuf-c protocomm pthread qr_code_ar remote_board remote_relay remote_relay_array request requests_ar sdmmc secboard soc spi spi_flash spiffs sys_control task_monitor_ar task_tracker tcp_transport telemetry uart_ar_esp32 ulp unity usb utils utils_ar vcnl4040 vfs w5100 watchdog_ar wear_levelling webserver_ar wifi_provisioning wpa_supplicant xtensa
-- Component paths: /project/components/access_request /esp-idf/components/app_trace /esp-idf/components/app_update /project/components/ar-access-req /project/components/ar-cfg-page /project/components/ar-config /project/components/ar-log /project/components/ar-mqtt /project/components/ar-wifi /project/components/ar_relay /project/components/ar_ulp_reset /esp-idf/components/bootloader /esp-idf/components/bootloader_support /project/components/bsp /esp-idf/components/bt /esp-idf/components/cmock /esp-idf/components/console /project/components/crypt_ar /esp-idf/components/cxx /esp-idf/components/driver /esp-idf/components/efuse /project/components/emergency_button /project/components/error_check_ar /esp-idf/components/esp-tls /esp-idf/components/esp_adc /esp-idf/components/esp_app_format /esp-idf/components/esp_coex /esp-idf/components/esp_common /esp-idf/components/esp_eth /esp-idf/components/esp_event /esp-idf/components/esp_gdbstub /esp-idf/components/esp_hid /esp-idf/components/esp_http_client /project/components/esp_http_client_ar /esp-idf/components/esp_http_server /esp-idf/components/esp_https_ota /esp-idf/c [... cut ...]
Project build complete. To flash, run this command:
/python_env/idf5.1_py3.11_env/bin/python ../esp_idf/esp-idf/components/esptool_py/esptool/esptool.py -p /dev/ttyUSB0 -b 921600 --before default_reset --after no_reset --chip esp32  write_flash --flash_mode dio --flash_size 4MB --flash_freq 80m 0x1000 build/bootloader/bootloader.bin 0xc000 build/partition_table/partition-table.bin 0x11000 build/ota_data_initial.bin 0x20000 build/ACCESSRUN_ATS_FIRMWARE.bin
or run 'idf.py -p /dev/ttyUSB0 flash'
```

</details>

esp-lrh, 2024-02-08:
Hi, @KaeLL 
This issue is caused by circular dependency between coexist lib and bluetooth lib.

Are you enable wifi and bluetooth in menuconfig, but only initialize wifi? (I can reproduce this issue in such scenarios) 

To fix this issue, you can
- Initialize bluetooth too. 

or

- Apply the following patch
[fix-circular-dependency-issue-on-esp32.PATCH](https://github.com/espressif/esp-idf/files/14204426/fix-circular-dependency-issue-on-esp32.PATCH)

We will fix this issue in subsequent versions, thanks for your report.

KaeLL, 2024-02-08:
> Are you enable wifi and bluetooth in menuconfig, but only initialize wifi? (I can reproduce this issue in such scenarios)

Yes.
> We will fix this issue in subsequent versions, thanks for your report.

You're welcome.

KaeLL, 2024-02-15:
@esp-lrh have you got an ETA for this? What about the backports?

esp-lrh, 2024-02-19:
Master branch need one or two days, backports need additional days.

RoKu-HTS, 2024-05-02:
> Hi, @KaeLL This issue is caused by circular dependency between coexist lib and bluetooth lib.
>
> Are you enable wifi and bluetooth in menuconfig, but only initialize wifi? (I can reproduce this issue [...]

Hey, What do you exactly mean by "Initialize Bluetooth too" what would be the minimal effort to achieve this?

esp-lrh, 2024-05-07:
Hi, @RoKu-HTS 
This issue has been fixed.

About "Initialize Bluetooth too":
The reason for this bug is that the user enable WIFI and BT in the sdkconfig, but did not call any BT functions, so the functions in the BT lib could not be found during linking coexist module. if user want to use BT, initialize BT will workaround this issue, if not, just disable BT in sdkconfig, this issue will not occur.