espressif/esp-idf #13115: Misleading description of uart_enable_tx_intr

ScumCoder, 2024-02-05:
https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/esp_driver_uart/include/driver/uart.h#L369

I couldn't find any mention of "TX_TIMEOUT" interrupt in ESP32 Technical Reference Manual.
Also, there seems to be no mention of this interrupt in the function body.
My guess is that someone copypasted the description of `uart_enable_rx_intr` and then replaced "Rx" with "Tx".

nopnop2002, 2024-02-06:
If you use the CTS signal, a transmission timeout occurs if CTS remains high.

ScumCoder, 2024-02-06:
> If you use the CTS signal, a transmission timeout occurs if CTS remains high.

Could you please provide the source of this statement? [Official manual](https://docs.espressif.com/projects/esp-idf/en/stable/esp32/api-reference/peripherals/uart.html) says nothing about this.

nopnop2002, 2024-02-06:
Pls search "hardware flow control" on this.
https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html

ScumCoder, 2024-02-06:
> Pls search "uart_wait_tx_done" on this. https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/uart.html

You literally posted a link to the same document as I did. As I said, it does not mention Tx timeouts.
Please either point to the concrete sentence in that document proving otherwise, or stop implying that I'm an idiot.

Moreover, I'm not talking about the documentation. I'm talking about the source code. Did you try reading the [body of the uart_enable_tx_intr function](https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/esp_driver_uart/src/uart.c#L596), as I suggested in the first post?

nopnop2002, 2024-02-06:
> As I said, it does not mention Tx timeouts.

There is a timeout description in uart_wait_tx_done.

ScumCoder, 2024-02-06:
> It seems that uart_enable_tx_intr() sets the wait time until UART_INTR_TXFIFO_EMPTY.

I see. Have a nice day.

closed 2025-03-08 by Alvin1Zhang