espressif/esp-idf #13116: Inconsistent behaviour of uart_set_rx_timeout

ScumCoder, 2024-02-05:
https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/esp_driver_uart/src/uart.c#L1775

Out of the three functions that set various thresholds - `uart_set_rx_full_threshold`, `uart_set_tx_empty_threshold` and `uart_set_rx_timeout` - the first two only do that after checking that the corresponding interrupt (`UART_INTR_RXFIFO_FULL` and `UART_INTR_TXFIFO_EMPTY` respectively) is set.
The latter, however, does not check `UART_INTR_RXFIFO_TOUT` and calls `uart_hal_set_rx_timeout` straight away.

nopnop2002, 2024-02-06:
Did you specify the correct value for tout_thresh?
```
tout_thresh -- This parameter defines timeout threshold in uart symbol periods. The maximum value of threshold is 126. tout_thresh = 1, defines TOUT interrupt timeout equal to transmission time of one symbol (~11 bit) on current baudrate. If the time is expired the UART_RXFIFO_TOUT_INT interrupt is triggered. If tout_thresh == 0, the TOUT feature is disabled.
```

ScumCoder, 2024-02-06:
> Did you specify the correct value for tout_thresh?

Not sure I understand your question... I did not specify anything, this post is just an observation I had while reading the IDF source code (see the linked line).

nopnop2002, 2024-02-06:
If you use uart_set_rx_timeout(), it will not work properly unless you specify the correct value for tout_thresh.