espressif/esp-idf #13117: Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.

jiapei100, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.1.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Successfully flashed.

### What is the actual behavior?

failed to flash with error:

```bash
[2/3] cd /opt/Espressif/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/Espressif/esp...es/get-started/hello_world/build -P /opt/Espressif/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyACM0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py vv4.8.dev1
Serial port /dev/ttyACM0
Connecting...

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):
  
  ~/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash ....../esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash 
cd /opt/Espressif/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/Espressif/esp-idf -D "SERIAL_TOOL=~/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=....../esp-idf/examples/get-started/hello_world/build -P /opt/Espressif/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the ....../esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_432108 and ....../esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_432108
```

### Steps to reproduce.

1. `idf.py build`
2. `idf.py flash`

### Build or installation Logs.

```plain
[2/3] cd /opt/Espressif/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/Espressif/esp...es/get-started/hello_world/build -P /opt/Espressif/esp-idf/components/esptool_py/run_serial_tool.cmake
esptool.py --chip esp32s3 -p /dev/ttyACM0 -b 115200 --before=default_reset --after=hard_reset write_flash --flash_mode dio --flash_freq 80m --flash_size 2MB 0x0 bootloader/bootloader.bin 0x10000 hello_world.bin 0x8000 partition_table/partition-table.bin
esptool.py vv4.8.dev1
Serial port /dev/ttyACM0
Connecting...

A serial exception error occurred: Write timeout
Note: This error originates from pySerial. It is likely not a problem with esptool, but with the hardware connection or drivers.
For troubleshooting steps visit: https://docs.espressif.com/projects/esptool/en/latest/troubleshooting.html
CMake Error at run_serial_tool.cmake:66 (message):
  
  ~/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3
  failed.

FAILED: CMakeFiles/flash ....../esp-idf/examples/get-started/hello_world/build/CMakeFiles/flash 
cd /opt/Espressif/esp-idf/components/esptool_py && /usr/bin/cmake -D IDF_PATH=/opt/Espressif/esp-idf -D "SERIAL_TOOL=~/.espressif/python_env/idf5.3_py3.10_env/bin/python;;/opt/Espressif/esp-idf/components/esptool_py/esptool/esptool.py;--chip;esp32s3" -D "SERIAL_TOOL_ARGS=--before=default_reset;--after=hard_reset;write_flash;@flash_args" -D WORKING_DIRECTORY=....../esp-idf/examples/get-started/hello_world/build -P /opt/Espressif/esp-idf/components/esptool_py/run_serial_tool.cmake
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the ....../esp-idf/examples/get-started/hello_world/build/log/idf_py_stderr_output_432108 and ....../esp-idf/examples/get-started/hello_world/build/log/idf_py_stdout_output_432108
```

### More Information.

- Ubuntu 22.04.3
- Python 3.10.2

dobairoland, 2024-02-06:
So, have you tried the linked troubleshooting guide?

dobairoland, 2024-02-08:
Hi @jiapei100. We won't be able to help you unless you provide us more information. You need to check your hardware as per suggestions in the troubleshooting guide and provide us with the results. This error in the early connection phase means that esptool (pyserial) cannot write to the port. Something must be wrong with the device driver, usb cable, ESP board (eg. power supply issue),...

closed 2024-02-19 by dobairoland