espressif/esp-idf #13118: ESP32-S3 dual CAN

GeorgeNegret, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

esp-idf : master 
device target : esp32s3
hardware : M5Stack CoreS3

I try to install the second twai driver with : 
`   twai_handle_t twai_bus_0;
    twai_handle_t twai_bus_1;
    #define TX_GPIO_NUM        GPIO_NUM_17 
    #define RX_GPIO_NUM        GPIO_NUM_18  
    #define TX2_GPIO_NUM      GPIO_NUM_9  
    #define RX2_GPIO_NUM      GPIO_NUM_8  

void can_init_twai() {
    twai_general_config_t g_config = TWAI_GENERAL_CONFIG_DEFAULT(TX_GPIO_NUM, RX_GPIO_NUM, TWAI_MODE_NORMAL);
    g_config.intr_flags |= ESP_INTR_FLAG_IRAM;
    g_config.rx_queue_len = 32;
    g_config.tx_queue_len = 32;
    twai_timing_config_t t_config = TWAI_TIMING_CONFIG_250KBITS();
    twai_filter_config_t f_config = TWAI_FILTER_CONFIG_ACCEPT_ALL();

    g_config.controller_id = 0;
    ESP_ERROR_CHECK(twai_driver_install_v2(&g_config, &t_config, &f_config, &twai_bus_0));
    ESP_LOGI(TWAI_TAG, "Driver 0 installed");
    ESP_ERROR_CHECK(twai_start_v2(twai_bus_0));
    ESP_LOGI(TWAI_TAG, "Driver 0 started");

    g_config.controller_id = 1;
    g_config.tx_io = TX2_GPIO_NUM;
    g_config.rx_io = RX2_GPIO_NUM;
    ESP_ERROR_CHECK(twai_driver_install_v2(&g_config, &t_config, &f_config, &twai_bus_1));
    ESP_ERROR_CHECK(twai_start_v2(twai_bus_1));
    ESP_LOGI(TWAI_TAG, "Driver 1 started");` 

but I have this error : 
(7178) gpio: GPIO[18]| InputEn: 1| OutputEn: 0| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(7178) gpio: GPIO[17]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 0| Pulldown: 0| Intr:0 
(7178) TWAI v2: Driver 0 installed
(7188) TWAI v2: Driver 0 started
    _ERROR_CHECK failed: esp_err_t 0x102 (ESP_ERR_INVALID_ARG) at 0x42017d3e
file: "./components/comp_twai/src/comp_twai.cpp" line 29
func: void can_init_twai()
expression: twai_driver_install_v2(&g_config, &t_config, &f_config, &twai_bus_1)

Can you help me ?  Thanks !

closed 2024-02-06 by GeorgeNegret

wanckl, 2024-02-07:
@GeorgeNegret  Sadly, S3 really have only one. C6 have 2, and coming P4 have 3, I guess P4 meet your high perform and more TWAI instant. Or use S3 plus C6 :face_in_clouds:

bobkim938, 2024-03-01:
@wanckl Is it possible to create two CAN nodes with one CAN controller?

wanckl, 2024-03-01:
@bobkim938 

What you mean a `BUS`, a physical H/L bus?

CAN is type of async communication, basically, different bus have only baud rate different.

TWAI peripheral self actually is able to config/switch baud rate dynamically, Of course, it have time cost and will loss data during switching.     

How ever the point is, if you connect different buses, one baud rate transfer will lead another baud rate node/devices into `BUS_ERROR`, then you need an extra `switch hardware` for H/L line or CAN TX/RX line.       

Or even simulate a `sw_can` by CPU software

Theoretically they are all doable, but no need to. So, impossible!

bobkim938, 2024-03-01:
@wanckl Thanks!!