espressif/esp-idf #13120: FreeRTOS Runtime Counter support.

ammaree, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1922-g5454d37d49

### Espressif SoC revision.

All ESP32 I believe

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

N/A

### Power Supply used.

USB

### What is the expected behavior?

Valid useable runtime information being returned.

### What is the actual behavior?

Runtime counter support is inconsistent and largely invalid.

### Steps to reproduce.

https://vscode.dev/github/espressif/esp-idf/blob/master/components/freertos/esp_additions/freertos_tasks_c_additions.h#L488
Should be using 'configRUN_TIME_COUNTER_TYPE' to support both u32_t and u64_t

https://vscode.dev/github/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h#L521-L522
We are using a 32 bit data source (with 12.5nSec resolution here) so the value will wrap every minute?

https://vscode.dev/github/espressif/esp-idf/blob/master/components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h#L523
We are using an int64_t data source (already not ideal, u64_t preferred) and then casting it to a uint32_t?

It is quite possible that there are more locations where configRUN_TIME_COUNTER_TYPE is not used, or where the source data is not valid.

o-marshmallow, 2024-02-07:
Hello @ammaree ,

Thank you for noticing this, we are going to look into this and fix it

ammaree, 2024-02-14:
@o-marshmallow 
The solution is very simple.

In components/freertos/esp_additions/freertos_tasks_c_additions.h:
```
488: - uint32_t ulRunTimeCounter;
488: + configRUN_TIME_COUNTER_TYPE ulRunTimeCounter;
```
in components/freertos/FreeRTOS-Kernel/portable/xtensa/include/freertos/portmacro.h:
```
523: - #define portALT_GET_RUN_TIME_COUNTER_VALUE(x) do {x = (uint32_t)esp_timer_get_time();} while(0)
523: + #define portALT_GET_RUN_TIME_COUNTER_VALUE(x) do {x = (configRUN_TIME_COUNTER_TYPE)esp_timer_get_time();} while(0)
```

Have tested this fix over last week on many different hardware platforms, works 100%

Maybe possible to get the fix through ASAP...

ammaree, 2024-04-05:
@o-marshmallow 

+99 for fixing this, only requires the 2 lines above to restore proper functioning.

Please

KonstantinKondrashov, 2024-04-09:
Hi @ammaree!
Thanks for the help with it. I created an MR with these changes.

closed 2024-04-23 by espressif-bot