espressif/esp-idf #13121: ESP-IDF rc1 5.2 problem in BT component header missing file, also cJson

filzek, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-rc1

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Missing "esp_bt_defs.h" file name found in the following component(s): bt(C:\Espressif\frameworks\esp-idf-v5.2\components\bt\host\bluedroid\api\include\api\esp_bt_defs.h). Maybe one of the components needs to add the missing header directory to INCLUDE_DIRS of idf_component_register call in CMakeLists.txt.

CMakeLists.txt.
idf_component_register(SRCS "app_main.c"
                       INCLUDE_DIRS ""
                       REQUIRES spi_flash esp_partition esp_wifi nvs_flash json driver bt
                       app_update esp_system wear_levelling fatfs littlefs
                       )

target_compile_options(${COMPONENT_LIB} PRIVATE -Wno-unused-const-variable)

even add the bt component it gives error using the header
#include "esp_bt_defs.h"

also the same using cJson, to solve I have add the json to the idf_component_register, missing this info in the manual at https://docs.espressif.com/projects/esp-idf/en/v5.2-rc1/esp32

### What is the actual behavior?

must build

### Steps to reproduce.

install, add the componente build incluing the header.

esp-cjh, 2024-02-07:
Hi @filzek What example are  you building? Could you please give me the commit ID of the ESP-IDF you using?

filzek, 2024-02-07:
HEAD is now at 11eaf41b37 Merge branch 'bugfix/fix_scan_request_ifs_error_v5.2' into 'release/v5.2'

filzek, 2024-02-27:
We manage to fix it, not sure if the BT include doesnt work even with the requires BT info, so, in a double development to have bluedroid and nimble options the new SDK can bring both headers, so, we did include it manually.

closed 2024-02-27 by filzek