espressif/esp-idf #13122: Clarefy how to view BLE services,characteristics and descriptors added dynamicly to see input parameters

DavidHemb, 2024-02-06:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Clarefy how to view BLE services,characteristics and descriptors added dynamicly to see input parameters. I have trubble location BLE service, etc in the code/repository. I assume it does not exist in the code since it is addad dynamicly but i am not finding some kind of CSV file or simular holding data for this. Again i want to see list of services, characteristics in said services and there discriptors also what is nessesary to send to the characteristic via gatt.
I might just be slow!
Thank you beforehand!

weiyuhannnn, 2024-02-07:
Hello @DavidHemb, Thanks for using our Espressif product! Could you please specify which chip you are using for BLE?

DavidHemb, 2024-02-07:
Hello @weiyuhannnn i use ESP32-S3

zhp0406, 2024-02-19:
@DavidHemb 
What specific content do you refer to as 'input parameters' here?
We support two sets of hosts: nimble host and bluedroid host.
In the bluedroid host, we demonstrate the creation of a server and other related content in the example below:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_server

If you wish to use the nimble host, you can refer to the example below:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble

zhp0406, 2024-02-19:
If you want to print the database, you can use the following API in the bluedroid host:
esp_ble_gatts_show_local_database in
https://github.com/espressif/esp-idf/blob/master/components/bt/host/bluedroid/api/esp_gatts_api.c#L429C1-L430C1

If you're using the nimble host, please use
ble_gatts_show_local.

DavidHemb, 2024-02-19:
I wish to use the WIFI config function you use in your test/demo app. I cannot find what format and what data to send to said function.

An example of this would be ssid and password in byte format

I just wish to read more where BLE sends to serverside. Landing function.

zhp0406, 2024-02-19:
In BLE GATT, there are two roles, namely server and client.
The server can send data to the client using notify and indicate.
such as esp_ble_gatts_send_indicate

The client can send data to the server using write,
such as **esp_ble_gattc_write_char**.

And can read data using read,
such as **esp_ble_gattc_read_char**.

zhp0406, 2024-02-19:
this is gatt client example:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_client

zhp0406, 2024-02-19:
You need to transmit the SSID and password information, which is a normal data flow for BLE. You can organize the data in your own way as long as the server can parse it. Additionally, if you are using security features, please refer to:

Server-side:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_server

Client-side:
https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/bluedroid/ble/gatt_security_client

DavidHemb, 2024-02-19:
I am sorry maybe i am not understanding you

In this app of yours https://github.com/espressif/esp-idf-provisioning-android

There is a function to simply set your ESP32 -S3 chips WiFi config to you own on you device.

I have found the method that sends this information here https://github.com/espressif/esp-idf/blob/4fb231088d97938e064c09ba5b37266515528eda/tools/esp_prov/transport/transport_ble.py

I only wish to see the method where it recives the BLE signal and sets WiFi config from said signal in this repository. 

My goal is to replicate this flow setting WiFi configuration from your app to your chip, but i am unable to find where in this repository BLE is recived and then set as WiFi configurations.

closed 2024-03-07 by DavidHemb