espressif/esp-idf #13127: [ESP_OTA] Give the possiblity to dissable `it->need_erase == 0` verification on `esp_ota_write_with_offset()`

mbastida123, 2024-02-07:
### Is your feature request related to a problem?

OTA update fails with:

```
it->need_erase == 0 && "must erase the partition before writing to it"
```

In this line: 
https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/app_update/esp_ota_ops.c#L273

I understand why this error appears and it is absolutely not a bug on either side. The problem is that with our use case this verification should not be there since we handle the deletion of the OTA sectors ourselves.

### Describe the solution you'd like.

To have a configuration option (via CONFIG_DISABLE_OTA_PARTITION_ERASE_CHECK for example) to disable this assert.

### Additional context.

I have succesfully implemented an OTA multisession update. Some context:

Sometimes our LTE connection is unstable and so the big .bin firmware file is a problem to download. 
To try to minimize this issue we have implemented the possibility to restart an OTA download at the point it became stuck the previous time. We do that by periodically saving the amount of bytes written into the partition on a NVS variable. This way on the second try the system checks wheter the previous download was interrumpted. If it was it makes another HTTP call with range-request enabled (https://developer.mozilla.org/en-US/docs/Web/HTTP/Range_requests) to only request the endpoint the part of the .bin we need. It then writes with ` esp_ota_write_with_offset()` the necessary data.

This means that not all the partition should cleared since we want to reuse the previous data written in it (to some extent). So this discards the option `OTA_SIZE_UNKNOWN` in `esp_ota_begin()`.
The other option would be `OTA_WITH_SEQUENTIAL_WRITES` but as the code states: 

> Used for esp_ota_begin() if new image size is unknown and erase can be done in incremental manner (assuming write operation is in continuous sequence) 

So this also discards this option.

Petzep, 2024-02-07:
If you set the image size to ```OTA_WITH_SEQUENTIAL_WRITES```, it will disable the check.

See:
https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/app_update/esp_ota_ops.c#L168

For my own code, I use:
```esp_ota_begin(otaPartition, OTA_WITH_SEQUENTIAL_WRITES, &updateHandle);```

mbastida123, 2024-02-07:
> If you set the image size to `OTA_WITH_SEQUENTIAL_WRITES`, it will disable the check.
>
> See: [...]

I think you are mistaken.

From your line if I set it to `OTA_WITH_SEQUENTIAL_WRITES` then `new_entry->need_erase=true`

Which means that 
```
assert(it->need_erase == 0 && "must erase the partition before writing to it");
```

Will convert to 
```
assert(1 == 0 && "must erase the partition before writing to it");
```

will fail, resulting in the error I was showing.

Am I right?

hmalpani, 2024-02-29:
Hello @mbastida123 
Does calling `esp_ota_begin()` with `OTA_SIZE_UNKNOWN` helps?

martibc16, 2024-03-15:
> Hello @mbastida123 Does calling `esp_ota_begin()` with `OTA_SIZE_UNKNOWN` helps?
> 

Hi @hmalpani and sorry for the late response.

`OTA_SIZE_UNKNOWN` won't work because according to the `esp_ota_begin()` begin method:
>         // If input image size is 0 or OTA_SIZE_UNKNOWN, erase entire partition

> Can you also explain the workflow for your OTA update? Are you calling esp_ota_begin() multiple times? Are you trying to perform OTA update across multiple reboots?

I only call `esp_ota_begin()` once. However, I wan't to perfom OTA across multiple reboots.
This is already achieved and working perfectly. However, I did have to make the patch to the ESP-IDF framework for it to work. Considering my use case (OTA accross multiple reboots) is something that could be expected there should be an option to disable the partition erase verification.

mbastida123, 2024-04-09:
Hi @hmalpani, any news on this?

hmalpani, 2024-04-22:
Hello @mbastida123 
As of now, ESP-IDF doesn't support OTA resumption i.e OTA over multiple reboots. I think OTA resumption will be a useful feature. We will discuss it internally on its implementation.

devprofile98, 2024-08-31:
Hi, is there any update on this feature request? we also need this feature and it is working with our patch to ESP-IDF locally right now, but it is not a permanent solution. so if there is any upcoming fix for this, we will be glad to wait for it, otherwise please let us know so we can find a permanent solution for it.
Thank you.

devprofile98, 2024-09-06:
@hmalpani hi, can you please answer my question above?

nileshkale123, 2024-09-06:
Hello @devprofile98 

We are actively working to provide full support for OTA resumption across multiple reboots. You can expect this feature to be available soon.

devprofile98, 2024-09-07:
Thank you so much!

i-e-b, 2025-01-17:
Any progress on this?

nileshkale123, 2025-01-20:
Hello @i-e-b ,

The MR regarding this will merge soon. We are working on final testing.
Thank you

i-e-b, 2025-01-21:
For anyone who finds this, if the official solution is not out yet; this is what I did:

Add this to ota_ops.c/h
```
esp_err_t esp_ota_write_skip(const esp_ota_handle_t handle, const size_t offset)
{
    // find ota handle in linked list
    for (ota_ops_entry_t* it = LIST_FIRST(&s_ota_ops_entries_head); it != nullptr; it = LIST_NEXT(it, entries)) {
        if (it->handle == handle) { // target partition
            if (it->wrote_size > offset) {
                ESP_LOGE(TAG, "OTA write skip target is before current offset (current = %lu, requested = %u)", it->wrote_size, offset);
                return ESP_ERR_OTA_PARTITION_CONFLICT;
            }

            it->wrote_size = offset;
            return ESP_OK;
        }
    }

    //if go to here ,means don't find the handle
    ESP_LOGE(TAG,"not found the handle");
    return ESP_ERR_INVALID_ARG;
}
```

- Store last written offset in NVS, along with the partition you're writing to.
- Pick partition for OTA with `esp_partition_t *p = esp_partition_find_first(ESP_PARTITION_TYPE_APP, myRecoverPartition, NULL);`
- After `esp_ota_begin(p, OTA_WITH_SEQUENTIAL_WRITES, &update_handle)`, but **before** `esp_ota_write(...)`, call `esp_ota_write_skip(update_handle, recoverOffset);`

closed 2025-01-26 by espressif-bot

devprofile98, 2025-01-26:
@i-e-b Hi, considering this issue closed, in which release should we expect this feature?

mahavirj, 2025-01-27:
@mbastida123 @devprofile98 @i-e-b @martibc16 

HTTPS OTA resumption feature is merged with https://github.com/espressif/esp-idf/commit/23ce1cc47d229dbe4f7c15cbbf2bdb91230403ee. This feature will be part of IDF 5.5 release. Please have a try and let us know if you have any feedback here.