espressif/esp-idf #13128: changing ADC .bitwidth, result in E (348) adc_oneshot: adc_oneshot_config_channel(138): invalid bitwidth ,, why is that ??

5ami, 2024-02-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

IDF V5.0

### Espressif SoC revision.

esp32s3  

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-s3-DevKitC-1 v1.1  

### Power Supply used.

USB

### What is the expected behavior?

Changing the bitwidth in 
 //-------------ADC1 Config---------------//
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = ADC_BITWIDTH_DEFAULT,
        .atten = EXAMPLE_ADC_ATTEN,
    };
Should result in ADC changing its bitwidth. 

### What is the actual behavior?

E (348) adc_oneshot: adc_oneshot_config_channel(138): invalid bitwidth

### Steps to reproduce.

1. Step
Get ADC oneshot example from ESP-IDF 
2. Step
Change  ADC_BITWIDTH_DEFAULT to another value in the enum adc_bitwidth_t ( for example **ADC_BITWIDTH_10**
 //-------------ADC1 Config---------------//
    adc_oneshot_chan_cfg_t config = {
        .bitwidth = ADC_BITWIDTH_DEFAULT,
        .atten = EXAMPLE_ADC_ATTEN,
    };
3. Step
Build and flash  == >> ESP32s3 Will keep restarting (E (348) adc_oneshot: adc_oneshot_config_channel(138): invalid bitwidth) 
...

nopnop2002, 2024-02-07:
As far as I remember, this issue was fixed in V5.1.

5ami, 2024-02-12:
I will see what  they did in IDF 5.1. Because for me it seemed that it is a hardware issue, the ADC hardware don't support changing bitwidth, maybe thats why they added a check to prevent it. also I have tested changing the register value using a LL function, and the system crashed as well

suda-morris, 2024-02-18:
Yes, this is a hardware limitation. 

https://github.com/espressif/esp-idf/blob/master/components/soc/esp32s3/include/soc/soc_caps.h#L113-L114

5ami, 2024-02-18:
Alright, Thanks, It would be nice if you could add it to the errata sheet.