espressif/esp-idf #13134: I²C NACK on write/address byte does not error using i2c_master

MatthiasKunnen, 2024-02-07:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

<sup>keywords: I²C NACK no error, i2c_master, i2c master, new driver<sup>

IDF version: 5454d37d496a8c58542eb450467471404c606501 (current master)

## Problem/Current behavior

When using the new I²C driver (`i2c_master`) functions, a NACK is perceived as an ACK on both slave
addresses and write bytes.

When the address byte results in a NACK (no device with this address is on the bus), the driver
continues as if an ACK was received.  
When a write byte results in a NACK, the driver continues as if an ACK was received.

This affects the following functions:
- `i2c_master_receive`
  - slave address ACK not verified
- `i2c_master_transmit`
  - slave address ACK not verified
  - write byte ACK not verified
- `i2c_master_transmit_receive`
  - slave address ACK not verified
  - write byte ACK not verified

A non-exhaustive list of things that could go wrong because of this:
- `i2c_master_transmit` on a missing slave appears to succeed.
- `i2c_master_transmit` on an existing but busy slave appears to succeed.
- `i2c_master_receive` and `i2c_master_transmit_receive` on a missing slave results in _fake_ `FF` bytes which is not distinguishable from an existing slave and valid `FF` bytes.

This is a regression from the old driver which verifies both the slave address ACK and the write byte's ACK, e.g.:
- `i2c_master_write_read_device`, address byte
  https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/driver/i2c/i2c.c#L1077
- `i2c_master_write_read_device`, registry address byte
  https://github.com/espressif/esp-idf/blob/5454d37d496a8c58542eb450467471404c606501/components/driver/i2c/i2c.c#L1082

I²C _Read Byte_ example using `i2c_master_transmit_receive` resulting in `ESP_OK` and `FF` output.
[image]

<details>
<summary>Code that generated this signal</summary>

```c
i2c_device_config_t dev_cfg = {
    .dev_addr_length = I2C_ADDR_BIT_LEN_7,
    .device_address = 0x2F,
    .scl_speed_hz = 50000,
};

i2c_master_dev_handle_t dev_handle;
ESP_ERROR_CHECK(i2c_master_bus_add_device(bus_handle, &dev_cfg, &dev_handle));

uint8_t readRegister = 0x20;
uint8_t receive[1]; // Will be 0xFF

 // Result will be 0, should be non-zero, e.g. 259
esp_err_t result = i2c_master_transmit_receive(dev_handle, &readRegister, 1, receive, 1, -1);
```

</details>

## Expected behavior
To be compliant with I²C and/or SMBus, the controller upon detection of the NACK condition must generate a STOP condition to abort the transfer.

## Solution
- Enable ACK verification on every write byte.
- Enable ACK verification on the address byte.

I will be making a PR that attempts to address both.

mythbuster5, 2024-02-08:
https://github.com/espressif/esp-idf/issues/13136#issuecomment-1933340595

jrahlf, 2024-02-08:
Can you clarify which IDF versions "new I2C driver" refers to?

KaeLL, 2024-02-08:
v5.2 onwards.

mythbuster5, 2024-04-22:
Solved

closed 2024-04-22 by mythbuster5