espressif/esp-idf #13138: Get MCLK in hfp_ag

smishra123, 2024-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am working with hfp_ag on LyraT 4.3 board to build a custom product. The system almost works. The only issue left is getting MCLK from BT controller to the ES8388 audio chip. I have configured the audio chip and verified the configuration. However ES8388 
 on LyraT 4.3 seems to need the MCLK to function. 

I added the following function calls in hfp_ag main.c adapted from i2s_check_set_mclk function in i2s_common.c. However this does not seem to work.  My assumption was that the Bluetooth stack and controller generated the I2S signals and were also generating the MCLK, but that the MCLK was not made available on the GPIO pin since that pin could vary from board to board.

    gpio_hal_iomux_func_sel(PERIPHS_IO_MUX_GPIO0_U, FUNC_GPIO0_CLK_OUT1);

    //gpio_ll_iomux_pin_ctrl(is_apll ? 0xFFF6 : (is_i2s0 ? 0xFFF0 : 0xFFFF));
    gpio_ll_iomux_pin_ctrl(0xFFFF); // i tried all 3 combinations

esp-qing, 2024-02-18:
Hi, this may be a configuration issue of the board in IDF. Please refer to the examples in ADF to modify the board configuration.
https://github.com/espressif/esp-adf/tree/master/examples/player/pipeline_hfp_stream

Alvin1Zhang, 2025-03-10:
Thanks for reporting, will cause due to inactivity, feel free to reopen with more updates.

closed 2025-03-10 by Alvin1Zhang