espressif/esp-idf #13140: OTA OVER BLE MESH

shreenidhi-ayla, 2024-02-08:
### Is your feature request related to a problem?

Hello there Espressif team!

We currently working with the ESP32-S3 in a BLE MESH app. I'm mostly using the Vendor model to transmit the values that I need, and it works as expected.

I would like to add an OTA feature. I would like to know which model can be used to for this? 

Thanks a lot in advance.

XionHsieh, 2024-02-21:
You can refer to here:

https://github.com/espressif/esp-iot-solution/tree/master/examples/bluetooth/ble_ota

shreenidhi-ayla, 2024-02-21:
Thanks you @XionHsieh

shreenidhi-ayla, 2024-03-08:
@xiewenxiang  will this OTA example work on BLE_mesh?