espressif/esp-idf #13143: Esp32-c6-wroom-1 some pins pulses after waking from sleep

Suxsem, 2024-02-08:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I'm building a zigbee sleepy device that wakes up every 7 seconds and light sleep for most of the time. 
For power consumption I'm using
CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP=y

Unfortunately there is a big issue: every time the board wakes up from light sleep, some pins, i.e. IO23 but I think all SDIO related pins, briefly pulses high, causing an unwanted interrupt in an attached external device.

I read from datasheet that IO23 at reset is WPU, so one would expect a brief high pulse when the chip is power up, but not every time it wakes up from light sleep. 

I really hope there is a solution for this problem because I already produced pcbs and cannot change pins assignment. 

Thank you

esp-wzh, 2024-02-18:
Hi, @Suxsem, if `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` is not enabled, will the problem still occur?

If `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP` is enabled, you need to call [gpio_hold_dis](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32c6/api-reference/peripherals/gpio.html#_CPPv413gpio_hold_dis10gpio_num_t) before each GPIO configuration of the GPIO and call  [gpio_hold_en](https://docs.espressif.com/projects/esp-idf/en/v5.2/esp32c6/api-reference/peripherals/gpio.html#_CPPv412gpio_hold_en10gpio_num_t) afterwards

Suxsem, 2024-02-18:
It works!! Thank you!

But why is that?

esp-wzh, 2024-02-18:
On the esp32c6 the GPIO belongs to the peripheral power domain, if you enable `CONFIG_PM_POWER_DOWN_PERIPHERAL_IN_LIGHT_SLEEP`, the peripheral power domain will be turned off during lightsleep, meaning that all configurations to the GPIO will be lost and the GPIO will not maintain its state during sleep. After waking up the esp32c6 will use the dedicated retention DMA to restore the GPIO configuration, so the state of the GPIO is only correct in the active state, if you need to maintain the GPIO state during sleep, you need to call the `gpio_hold_en` to latch the configuration on the GPIO PADs

Suxsem, 2024-02-18:
Understood, thank you. Maybe you could document it somewhere. Can be closed

closed 2024-02-28 by espressif-bot