espressif/esp-idf #13144: TRM contains incorrect values for lowest LEDC duty resolution

ScumCoder, 2024-02-08:
v5.0 ESP32 Technical Reference Manual has the following table on p. 387:

[image]

The right column of this table is wrong.
If one were to feed, say,  80 MHz clock frequency, 1 kHz PWM frequency, and 6 resolution bits into the `ledc_calculate_divisor` function
https://github.com/espressif/esp-idf/blob/482a8fb2d78e3b58eb21b26da8a5bedf90623213/components/driver/ledc/ledc.c#L301
the result would be
```
( ( (uint64_t) 80000000 << 8 ) + ((1000 * 64) / 2 ) )
           / (1000 * 64)
= 320000,5
```
which is more than 0x3FFFF and thus fails `LEDC_IS_DIV_INVALID`, leading to the `requested frequency and duty resolution can not be achieved`  error.

Following [this post](https://esp32.com/viewtopic.php?t=6701#p28822), the proper formula in case of the lowest resolution should be not
```
integer (log 2 (LEDC_APB_CLK / frequency))
```
but rather
```
integer (ceil(log 2 (LEDC_APB_CLK / frequency)))
```
Thus, minimum resolution for APB_CLK and 1 kHz is not 6 bits but 7, for 5 kHz it's not 3 bits but 4, etc.

espwangning, 2024-02-22:
Thanks for the detailed description!

You are correct that the minimum resolution should be rounded up instead of rounded down. We will update the all the corresponding TRM in the next release.

ESP-Marius, 2024-04-18:
Seems like the updated TRM has been released now, so i'll close this issue.

Feel to reopen if you think there are still any outstanding issues.

closed 2024-04-18 by ESP-Marius