espressif/esp-idf #13147: How can I modify Thread dBm?

i-kratko, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hello, I am conducting some experiments using thread and I have run into the following issue:
The LDO I have chosen to use for my project cannot handle the default "txpower" of the esp32c6 (305mAh with 20dBm), however it works just fine with 12dBm cause its drawing around 190 mAh. The problem is the following: currently the only way I have discovered that you can modify this value is by using the "txpower <desired dbm>" in the openthread cli. Is there a way I can hardcode the value so the esp boots up with it everytime?

chshu, 2024-02-19:
The default txpower could be changed by calling the `otPlatRadioSetTransmitPower()` API.

i-kratko, 2024-02-20:
> The default txpower could be changed by calling the `otPlatRadioSetTransmitPower()` API.

So in my case I should use otPlatRadioSetTransmitPower(12) somewhere before or after the initialisation of the network interface?

chshu, 2024-02-20:
Call `otPlatRadioSetTransmitPower(esp_openthread_get_instance(), 10);` after the `esp_openthread_init()`.

closed 2024-02-23 by chshu