espressif/esp-idf #13149: Definition of the PACKED_ATTR has a semicolon at the end of definition which limits flexibility in usage

slapab-vammo, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The `PACKED_ATTR` has been introduced in the v5.2 ESP-IDF version as macro defined in the `esp-idf/components/esp_common/include/esp_attr.h` and looks like this:   
```c
// Forces the data to be tightly packed with minimum required padding and no extra bytes are added for alignment
#define PACKED_ATTR __attribute__((packed));
```
Definition of this macro has a semicolon at the end of the line which introduces limitation in flexibility of usage.
For example consider this code: 
```c
typedef struct PACKED_ATTR {
    int field1;
    int field2;
} my_type_t;
```

Current format introduces a `;` before `{` thus code is ill formed and we have a compilation errors.  

--- 

### Affected versions 

This issue exists also in the latest (for this time) release `v5.2-rc1`.

---

### Proposed resolution
The proposed resolution is to just remove a `;` from the `PACKED_ATTR` definition.

SoucheSouche, 2024-02-12:
@slapab-vammo,
Thank you for reporting this issue. I can see the semicolon was added recently but I can't see any reason why. I will remove it from the macro.

AxelLin, 2024-03-05:
The fix is already in v5.2 branch, please close this issue. @SoucheSouche

closed 2024-03-05 by SoucheSouche