espressif/esp-idf #13150: esp_tinyuf2 and esp_tinyusb in the same project

kaspernyhus, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Use esp_tinyuf2 DFU along features from the esp_tinyusb component

### What is the actual behavior?

not compiling

### Steps to reproduce.

add to idf-component.yml

```
## IDF Component Manager Manifest File
dependencies:
  espressif/esp_tinyuf2: "^0.2.0"
  espressif/esp_tinyusb: "^1.4.3"
```

tore-espressif, 2024-02-13:
Hello @kaspernyhus thank you for the issue.

That is a very reasonable requirement, however, at the moment these two components do not work together.
What specific feature from `esp_tinyusb` do you need? It should be feasible to merge features from `esp_tinyusb` to `esp_tinyuf2` as they are both based on TinyUSB

kaspernyhus, 2024-02-13:
Hi Tomas, thanks for getting back.

I need HID and USB Audio which I pulled into `esp_tinyusb` in a fork of `idf-extra-components` [here](https://github.com/bang-olufsen/idf-extra-components/tree/kank/usb-audio/usb/esp_tinyusb)

Which approach would you recommend:
1. merge the features (hid and audio) from `esp_tinyusb` into `esp_tinyuf2`
2. pick UF2 functionality from `esp_tinyuf2` and build on top of MSC in `esp_tinyusb`

thanks a lot!

leeebo, 2024-02-15:
```
Is it possible to have these features at the same time?
```

@kaspernyhus Yes, I recommend you pick the esp_tinyuf2 code to [your tinyusb project](https://github.com/bang-olufsen/idf-extra-components/tree/kank/usb-audio/usb/esp_tinyusb), just like what you have done for adding UAC class to `esp_tinyusb`. (add uf2 related source files and modify the descriptors and so on.)

```
Booting between modes for UF2/non-UF2 is fine
```
If you have enough flash space, it's a good idea to build a separate UF2-enabled APP then flash to a separate partition, in this case you don't need to merge the code. please refer [boot-from-test-firmware](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-guides/bootloader.html#boot-from-test-firmware) and the [esp-box chatgpt demo](https://github.com/espressif/esp-box/tree/master/examples/chatgpt_demo)

StevenMacias, 2024-02-19:
> ```
> Is it possible to have these features at the same time?
> ```
>
> @kaspernyhus Yes, I recommend you pick the esp_tinyuf2 code to [your tinyusb [...]

@leeebo thank you so much for your answer. I have checked the boot-from-test-firmware documentation. Is it only possible to boot from a test partition by using a GPIO pin? Is there any method that can be used to software reboot in a different partition? Thanks again!

leeebo, 2024-02-20:
@StevenMacias Sure, you can use API `esp_ota_set_boot_partition` to choose boot partition, please refer https://github.com/espressif/esp-box/blob/master/examples/chatgpt_demo/factory_nvs/main/main.c#L41C5-L41C31

StevenMacias, 2024-02-21:
@leeebo Thanks again for the fast and useful reply. Currently we are using the `idf.py uf2-ota` command to generate the `.uf2` files. How we could include the `UF2-enabled APP` in the `.uf2` firmware? We are interested on separating the stacks as you suggested but also on being able to update both the main application and the UF2-enabled APP.

leeebo, 2024-04-22:
@StevenMacias If using `UF2-enabled APP` + `normal uf2 app`, the `UF2-enabled APP` works as the special featured app / "bootloader", if you want to update it, an additional partition "UF2-enabled APP" needs to be add.

Then you have to judge by name, and choose corresponding partition to write during the OTA process.

kaspernyhus, 2024-04-23:
Thanks for the help, we went the simple route with one uf2 partition. Will close this.

closed 2024-04-23 by kaspernyhus