espressif/esp-idf #13151: SPI linking problem.

ammaree, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1978-g4fb231088d

### Operating System used.

macOS

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Have a build system that, using parameters to select the devkit/config/libraries, will build firmware for up to 10 different hardware platforms and about 40 major configs.

Expect the firmware to compile and link successfully.

### What is the actual behavior?

The system works for all the devkits except 1 (ESP-WROVER-KIT v4.1) and also only when building with a GUI using the SPI interface with LEDC support for backlight. I have reduced the high level CMakeLists.txt file to a minimum just for this devkit and the symptom remains exactly the same. ALL references to led_hal_* and spi_hal_* symbols remain unresolved. 

### Steps to reproduce.

# ESP-IDF top level cmake file
cmake_minimum_required(VERSION 3.5)
################################# Project Version Support
set( Vmajor  0 )							# Major
set( Vminor  6 )							# Minor
set( Vpatch  0 )							# Patch
set( Vtweak  50 )							# Tweak
set( PROJECT_VER ${Vmajor}.${Vminor}.${Vpatch}.${Vtweak} )
################################ Build Definitions 
set( PHome /Users/andremaree )
set( PWork ${PHome}/DevSpace )
set( PComp ${PWork}/z-comp )
set( PRepo ${PWork}/z-repo )
set( P_SDK ${PWork}/z-sdk/esp-idf )
set( PExtn ${P_SDK}/examples/common_components/protocol_examples_common )
list( APPEND EXTRA_COMPONENT_DIRS ${PComp} )
list( APPEND EXTRA_COMPONENT_DIRS ${PComp}/devices )
list( APPEND EXTRA_COMPONENT_DIRS ${PExtn} )
############################## Ensure defaults set 
set( cmakeMODEL dk41 )
set( cmakeAEP 2 )
set( cmakeBUILD dbg )
set( cmakeGUI 2 )
set( EXTRA_REQUIRES "onewire esp_wrover_kit gui_basic ili9341 lcdgfx" )
set( cmakeTARGET esp32 )
list( APPEND EXTRA_COMPONENT_DIRS ${PRepo}/lcdgfx )
set( COMPONENTS "${COMPONENTS} driver espcoredump esptool_py esp_netif nvs_flash" )
set( COMPONENTS "${COMPONENTS} esp_driver_ledc esp_driver_spi hal" )
set( COMPONENTS "${COMPONENTS} buffers commands common hal_esp32 printfx rtos-support socketsX syslog systiming" )
set( COMPONENTS "${COMPONENTS} main rules" )
set( COMPONENTS "${COMPONENTS} ${EXTRA_REQUIRES}" )

set( ENV{IDF_PATH} ${P_SDK} )
set( ENV{cmakeMODEL} ${cmakeMODEL} )
set( ENV{IDF_TARGET} ${cmakeTARGET} )
include( ${P_SDK}/tools/cmake/project.cmake )
[... 82 lines cut ...]
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.1.literal+0x0): undefined reference to `ledc_hal_set_hpoint'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.1.literal+0x4): undefined reference to `ledc_hal_set_duty_int_part'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.1.literal+0x8): undefined reference to `ledc_hal_set_fade_param'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.3.literal+0x18): undefined reference to `ledc_hal_get_fade_end_intr_status'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.3.literal+0x1c): undefined reference to `ledc_hal_clear_fade_end_intr_status'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.3.literal+0x20): undefined reference to `ledc_hal_get_duty'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj):(.iram1.3.literal+0x24): undefined reference to `ledc_hal_set_duty_start'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_ls_channel_update':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:127:(.iram1.0+0xe): undefined reference to `ledc_hal_ls_channel_update'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `_ledc_update_duty':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:723:(.text._ledc_update_duty+0x30): undefined reference to `ledc_hal_set_duty_start'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_speed_mode_ctx_create':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:306:(.text.ledc_speed_mode_ctx_create+0x40): undefined reference to `ledc_hal_init'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_duty_config':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:227:(.iram1.1+0x1a): undefined reference to `ledc_hal_set_hpoint'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:230:(.iram1.1+0x2c): undefined reference to `ledc_hal_set_duty_int_part'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:232:(.iram1.1+0x42): undefined reference to `ledc_hal_set_fade_param'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_fade_isr':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:926:(.iram1.3+0x17): undefined reference to `ledc_hal_get_fade_end_intr_status'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_calc_fade_end_channel':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:909:(.iram1.3+0x60): undefined reference to `ledc_hal_clear_fade_end_intr_status'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: esp-idf/esp_driver_ledc/libesp_driver_ledc.a(ledc.c.obj): in function `ledc_fade_isr':
/Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:960:(.iram1.3+0xc6): undefined reference to `ledc_hal_get_duty'
/Users/andremaree/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/../lib/gcc/xtensa-esp-elf/13.2.0/../../../../xtensa-esp-elf/bin/ld: /Users/andremaree/DevSpace/z-sdk/esp-idf.v5x/components/esp_driver_ledc/src/ledc.c:1000:(.iram1.3+0x1bf): undefined reference to `ledc_hal_set_duty_start'
```

ammaree, 2024-02-09:
If it helps, I have checked and 
build/esp-idf/esp_driver_ledc/libesp_driver_ledc.a
build/esp-idf/esp_driver_spi/libesp_driver_spi.a
are present

adokitkat, 2024-02-12:
Hi. I think you shared your private keys and tokens here in the description (Google, Eureka ?). 

It is true `libesp_driver_ledc.a` and `libesp_driver_spi.a` are present - the linker error is from these files, they are calling functions from other files (like `ledc_hal_set_duty_start` -> `components/hal/include/hal/ledc_hal.h`, `components/hal/ledc_hal_iram.c`) which couldn't be found.
You can try adding `driver` and `hal` to your `REQUIRES` or `PRIV_REQUIRES`.

wanckl, 2024-02-19:
@ammaree    

I have a try on your top cmake file with  a little "IDF_ROOT_PATH" changes, but I can't find your issue.    
I think it maybe permission issue or cmake version issue, not very familiar with OSX, I'm using cmake 3.16.3, what your accurate cmake version.   

But easiest way I think you can build SPI and LEDC example to see if IDF is OK.

ammaree, 2024-02-22:
@adokitkat @igrr @wanckl 

I have been able to make the problem disappear but I still have no idea what caused it or why it happened.

In short, what I did was:
1. Organise all components into three categories being 1) application related; 2) ESPxx private HAL related; and 3) devices, having both application and HAL related interfaces.
2. 90% of all REQUIRES and PRIV_REQUIRES were removed from all components.
3. Application components were given REQUIRES "main", HAL components REQUIRES "esp32_hal" and devices were given both "main esp32_hal"
4. ALL application components are listed as REQUIRES in the CMakeLists.txt for component "main"
5. ALL private HAL components are listed as REQUIRES in the CMAKELists.txt for component "hal_esp32"
6. Very few components were given 1 or maximum 2 additional REQUIRES
7. The top level project CMakeLists.txt have only COMPONENTS "main"

It is the second time in the last 3 or 4 years that I have been confronted with EXACTLY the same problem.
In both cases the link problems were related to specifically the SPI related components in ESP-IDF.
In each case I wasted probably a full week first trying to identify and fix the cause but afterwards being forced to wipe all [PRIV_]REQUIRES and start from scratch.

I now only hope that this approach will be successful in future