espressif/esp-idf #13152: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or director

diplfranzhoepfinger, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or director

```
franz@franz-Ubuntu:~/master/esp-idf$ . ./export.sh
Detecting the Python interpreter
Checking "python3" ...
Python 3.11.6
"python3" has been detected
Checking Python compatibility
Checking other ESP-IDF version.
Not using an unsupported version of tool cmake found in PATH: 3.27.4. To use it, run '/usr/bin/python3 /home/franz/master/esp-idf/tools/idf_tools.py export --prefer-system'
ERROR: tool qemu-xtensa version esp_develop_8.1.3_20231206 is installed, but returned non-zero exit code (127) with error message:
/home/franz/.espressif/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin/qemu-system-xtensa: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory

ERROR: tool qemu-riscv32 version esp_develop_8.1.3_20231206 is installed, but returned non-zero exit code (127) with error message:
/home/franz/.espressif/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin/qemu-system-riscv32: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory

franz@franz-Ubuntu:~/master/esp-idf$ 

```

GTCLive, 2024-02-15:
This is so odd. A full week went by and the linux & plugin ide development environment remain broken without a single follow-up, acknowledgement and/or neither a complain. Makes you wonder if anyone dev espressif products on the linux side, so odd and worrying...

GTCLive, 2024-02-16:
Updating master and installing libslirp0 seems to resolve both the dependency and qemu error. Grateful for passing on the dockerfile, @igrr. Great pointers!

diplfranzhoepfinger, 2024-02-20:
```
sudo apt install libslirp0
```

diplfranzhoepfinger, 2024-02-20:
espressif/idf-eclipse-plugin#898

diplfranzhoepfinger, 2024-02-20:
```
OK

Executing /usr/bin/python3 /home/franz/esp-idf-v5.2/tools/idf_tools.py export --format=key-value
Not using an unsupported version of tool cmake found in PATH: 3.27.4. To use it, run '/usr/bin/python3 /home/franz/esp-idf-v5.2/tools/idf_tools.py export --prefer-system'
WARNING: tool qemu-xtensa version esp_develop_8.1.3_20231206 is installed, but returned non-zero exit code (127) with error message:
/home/franz/.espressif/tools/qemu-xtensa/esp_develop_8.1.3_20231206/qemu/bin/qemu-system-xtensa: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory

WARNING: tool qemu-riscv32 version esp_develop_8.1.3_20231206 is installed, but returned non-zero exit code (127) with error message:
/home/franz/.espressif/tools/qemu-riscv32/esp_develop_8.1.3_20231206/qemu/bin/qemu-system-riscv32: error while loading shared libraries: libSDL2-2.0.so.0: cannot open shared object file: No such file or directory

OPENOCD_SCRIPTS=/home/franz/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/share/openocd/scripts
ESP_ROM_ELF_DIR=/home/franz/.espressif/tools/esp-rom-elfs/20230320/
IDF_PYTHON_ENV_PATH=/home/franz/.espressif/python_env/idf5.2_py3.11_env
ESP_IDF_VERSION=5.2
PATH=/home/franz/.espressif/tools/xtensa-esp-elf-gdb/12.1_20231023/xtensa-esp-elf-gdb/bin:/home/franz/.espressif/tools/riscv32-esp-elf-gdb/12.1_20231023/riscv32-esp-elf-gdb/bin:/home/franz/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/franz/.espressif/tools/esp-clang/16.0.1-fe4f10a809/esp-clang/bin:/home/franz/.espressif/tools/riscv32-esp-elf/esp-13.2.0_20230928/riscv32-esp-elf/bin:/home/franz/.espressif/tools/esp32ulp-elf/2.35_20220830/esp32ulp-elf/bin:/home/franz/.espressif/tools/cmake/3.24.0/bin:/home/franz/.espressif/tools/openocd-esp32/v0.12.0-esp32-20230921/openocd-esp32/bin:/home/franz/.espressif/tools/ninja/1.11.1/:/home/franz/.espressif/python_env/idf5.2_py3.11_env/bin:/home/franz/esp-idf-v5.2/tools:$PATH
IDF_DEACTIVATE_FILE_PATH=/tmp/tmpgg6095nyidf_377683
Configured CDT Build Environment variables. You can check Preferences > C/C++ > Build > Environment
```

diplfranzhoepfinger, 2024-06-11:
https://github.com/espressif/esp-idf/pull/13949