espressif/esp-idf #13153: [Question] Are the idf tools expected to be pinned or not?

amrsoll, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Today, when running `. ~/esp-idf/export.sh` I get this error message

```
ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/usr/bin/python3 /home/alex/esp-idf-master/tools/idf_tools.py install' to install it.
Not using an unsupported version of tool openocd-esp32 found in PATH: 0.11.0. To use it, run '/usr/bin/python3 /home/alex/esp-idf-master/tools/idf_tools.py export --prefer-system'
```

It didn't use to happen, and I have not changed anything to the esp-idf repo, nor have I explicitely (on my own) made changes in `~/.espressif/`

It seems that I cannot expect the idf toolchain to always work on a pinned version. Is `export.sh` doing some auto-updates? If so, that is undesirable for multiple reasons
- It is slow, especially on slow connections (I will do `. export.sh` multiple times a day)
- Difficult to reproduce errors
- I might run into issues like the one I am describing.

If there is some behind the scenes changes to the resources in `.espressif`, could you let me know of a "read-only" version of `export.sh` ?

Same question goes for the toolchain we install using `install.sh`. If my repo `esp-idf` is pinned at tag `v5.1`, can I expect `install.sh` to install the exact same toolchain at any point in the future?

This is a follow up on a discussion I have had in #12539, where I explain that I have experienced oddities in the toolchain setup.

dobairoland, 2024-02-09:
Export doesn't do any updates.

Install does some (Python packages which are not pinned are updated). However, toolchains, debuggers and similar you are referring to are not updated unless you check out a different version of ESP-IDF.

dobairoland, 2024-02-09:
> ERROR: tool xtensa-esp32s2-elf has no installed versions. Please run '/usr/bin/python3 /home/alex/esp-idf-master/tools/idf_tools.py install' to install it.

This might happen if you remove those tools accidentally, or change the tool directory by an environment variable.

closed 2024-02-09 by dobairoland