espressif/esp-idf #13154: esp_a2d_sink_connect takes a long time to fail out sometimes if HCI_ERR_CONNECTION_EXISTS

ftab, 2024-02-09:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2-691-g7380f96017

### Espressif SoC revision.

ESP32-D0WD rev 1

### Operating System used.

macOS

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

ESP32-LyraT v4.3

### Power Supply used.

USB

### What is the expected behavior?

Connect to my device or give an error message right away

### What is the actual behavior?

Takes a minute to fail out sometimes

### Steps to reproduce.

I tested with LyraT v4.3 board and an iPhone 15 Pro

I modified the pipeline_a2dp_sink_and_hfp example in esp-adf to have a connect loop that calls esp_a2d_sink_connect

From scratch without any existing copy of esp-adf:
1. `cd ~/esp`
2. `git clone https://github.com/radiosound-com/esp-adf -b esp-bt-connect-bug --recursive`
3. `esp-adf/esp-idf/install.sh`
4. `export ADF_PATH=~/esp/esp-adf`
5. `source esp-adf/esp-idf/export.sh`
6. `cd esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp`
7. Modify line 369 of main/pipeline_a2dp_sink_and_hfp_example.c so that your device's address is set in the array (otherwise this will not connect to anything). Make sure your phone is in Bluetooth settings to get the pair prompt, if this device is not already paired to the LyraT.
8. `idf.py menuconfig` and set flash size to 4 MB or larger, set Bluetooth debug logs on
9. `idf.py build flash monitor`
10. After the device is connected, hit the Reset button on LyraT. Repeat step 10

If you have esp-adf and esp-idf already set up:
1. `cd esp-adf/examples/get-started/pipeline_a2dp_sink_and_hfp`
2. Use [this copy of pipeline_a2dp_sink_and_hfp_example.c](https://github.com/radiosound-com/esp-adf/blob/esp-bt-connect-bug/examples/get-started/pipeline_a2dp_sink_and_hfp/main/a2dp_sink_and_hfp_example.c) but modify line 369 so that your device's address is set in the array (otherwise this will not connect to anything). Make sure your phone is in Bluetooth settings to get the pair prompt, if this device is not already paired to the LyraT.
3. `idf.py menuconfig` and set flash size to 4 MB or larger, set Bluetooth debug logs on
4. `idf.py build flash monitor`
5. After the device is connected, hit the Reset button on LyraT. Repeat step 5

### Debug Logs.

```plain
[... 181 lines cut ...]

I (904) AUDIO_ELEMENT: [i2s_r] AEL_MSG_CMD_RESUME,state:1
I (904) AUDIO_PIPELINE: Pipeline started
I (904) BLUETOOTH_EXAMPLE: [ 7 ] Listen for all pipeline events
I (914) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
W (914) BT_APPL: reset flags
W (924) PERIPH_TOUCH: _touch_init
W (2354) BT_HCI: hcif conn complete: hdl 0x81, st 0xb
I (5924) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (10934) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (15944) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (20954) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (25964) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (30974) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (35984) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (40994) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (46004) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (51014) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
I (56024) BLUETOOTH_EXAMPLE: [ * ] Connect to my device
W (60914) BT_BTC: BTA_AV_OPEN_EVT::FAILED status: 2
```

### More Information.

Some resets it will reconnect after a few seconds, sometimes it will take a minute before reporting `BTA_AV_OPEN_EVT::FAILED status: 2` and allowing another attempt to connect. The latter is the problem. It is annoying to the end user if the device resets and then takes more than a minute to reconnect. This happens up to 50% of the time in our tests so far (on our custom board, but with the same connection attempt method I added to this example)

shangke1112, 2024-02-19:
**W (2354) BT_HCI: hcif conn complete: hdl 0x81, st 0xb**

The reason for this problem is that the mobile phone rejected the connection of esp32 through LMP_NOT_ACCEPT. The error code is 0xb (connection already exists). It is speculated that the restart speed of esp32 is too fast, and the mobile phone has not yet disconnected; Then ESP32 initiated the connection again and was rejected by the phone. 
There is no good solution, I think you can try delaying a period of time before connecting.

ftab, 2024-02-19:
> try delaying a period of time before connecting.

@shangke1112 I added a five second delay before trying to connect initially, and it seems to be connecting more often, but then that means there's a longer wait at startup before it's connected in all cases, not just when restarting. This may be an acceptable workaround for now. But if I can avoid that 5 second delay I'd prefer to. It seems like the status response HCI_ERR_CONNECTION_EXISTS should cause the esp_a2d_sink_connect to fail right away, instead it takes 60 seconds. Is there no way to cancel that?

shangke1112, 2024-02-20:
I suggest  that you can add a reset flag to control whether there is a delay .
### It seems like the status response HCI_ERR_CONNECTION_EXISTS should cause the esp_a2d_sink_connect to fail right away, instead it takes 60 seconds. Is there no way to cancel that?
We will confirm how to solve this problem later.