espressif/esp-idf #13156: event documentation misses semicolon ; after ESP_EVENT_DECLARE_BASE(EVENT_BASE) causing compile error when copy pasting

flyinggorilla, 2024-02-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

it is just confusing when the documentation misses the semicolon. 

### What is the actual behavior?

compile error on subsequent statement

### Steps to reproduce.

copy paste from documentation of 
ESP_EVENT_DECLARE_BASE(EVENT_BASE)

flyinggorilla, 2024-02-10:
sorry, super simple fix. just add the semicolon to the examples of 
ESP_EVENT_DECLARE_BASE(EVENT_BASE)

https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/peripherals/gptimer.html?highlight=timer#gptimer-register-event-callbacks

adokitkat, 2024-02-11:
Hi, can you point to which code example has this missing semicolon please? You can just copy-paste it here. Thank you.

flyinggorilla, 2024-02-12:
https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/esp_event.html?highlight=event%20loop#c.ESP_EVENT_ANY_BASE

Declaring and Defining Events[](https://docs.espressif.com/projects/esp-idf/en/latest/esp32s3/api-reference/system/esp_event.html?highlight=event%20loop#declaring-and-defining-events)
As mentioned previously, events consist of two-part identifiers: the event base and the event ID. The event base identifies an independent group of events; the event ID identifies the event within that group. Think of the event base and event ID as a person's last name and first name, respectively. A last name identifies a family, and the first name identifies a person within that family.

The event loop library provides macros to declare and define the event base easily.

Event base declaration:

ESP_EVENT_DECLARE_BASE(EVENT_BASE)   **;**  <---- MISSING semicolon
Event base definition:

ESP_EVENT_DEFINE_BASE(EVENT_BASE)   **;** <----- MISSING semicolon

thx

adokitkat, 2024-02-12:
I see now. However I don't think that's a mistake, since `ESP_EVENT_DECLARE_BASE` and `ESP_EVENT_DEFINE_BASE` are C macros which output statements. And in C after statement there needs to be a semicolon.

```c
#define ESP_EVENT_DECLARE_BASE(id) extern esp_event_base_t const id
#define ESP_EVENT_DEFINE_BASE(id) esp_event_base_t const id = #id
```

 You aren't supposed to copy it from docs like `ESP_EVENT_DECLARE_BASE(EVENT_BASE)`, you need to use it like e.g.:
 
```c
...
ESP_EVENT_DECLARE_BASE(s_your_base_variable);
...
 ```

closed 2024-02-20 by espressif-bot