espressif/esp-idf #13157: ESP32-C3 disconnecting after starting Wi-Fi

weiying-chen, 2024-02-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

My ESP32-C3 disconnects when I add this Rust code (note: there are no errors whatsoever):

    use embedded_svc::wifi::{ClientConfiguration, Configuration, Wifi};

    let mut wifi_driver = EspWifi::new(peripherals.modem, sys_loop, Some(nvs)).unwrap();
    wifi_driver.start().unwrap(); // more specifically, this line

This is what espmonior sees before my ESP32-C3 disconnects from my mini PC:

```
I (540) wifi_init: rx ba win: 6
I (550) wifi_init: tcpip mbox: 32
I (550) wifi_init: udp mbox: 6
I (550) wifi_init: tcp mbox: 6
I (560) wifi_init: tcp tx win: 5744
I (560) wifi_init: tcp rx win: 5744
I (570) wifi_init: tcp mss: 1440
I (570) wifi_init: WiFi IRAM OP enabled
I (570) wifi_init: WiFi RX IRAM OP enabled
Device disconnected; exiting
```

[This](https://gist.github.com/rajeshpachaikani/2ef7c11b4d7e8a6704dba0f323bcdd6d) is the full code.

Also [posted](https://www.reddit.com/r/embedded/comments/1andhul/esp32c3_disconnecting_after_starting_wifi/) on Reddit

MaxwellAlan, 2024-02-19:
Hi @weiying-chen 

`Device disconnected; exiting` this log seems from esp-rs framework, suggest move this issue to esp-rs

Thanks.

weiying-chen, 2024-02-19:
@MaxwellAlan The issue was that my ESP32-C3 was underpowered. Sorry for opening the issue.

closed 2024-02-19 by Alvin1Zhang