espressif/esp-idf #13158: Nimble: Compile error w/ -O2 optimization

mickeyl, 2024-02-10:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2-rc1-4-gcc649ea869

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Should compile without errors.

### What is the actual behavior?

Compile errors out.

### Steps to reproduce.

Compile.

### Build or installation Logs.

```plain
Executing action: all (aliases: build)
Running cmake in directory /home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/build
Executing "cmake -G Ninja -DPYTHON_DEPS_CHECKED=1 -DPYTHON=/home/mickey/.espressif/python_env/idf5.2_py3.10_env/bin/python -DESP_PLATFORM=1 -DCCACHE_ENABLE=0 /home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit"...
-- Computed build version as '139'
-- IDF_TARGET is not set, guessed 'esp32s3' from sdkconfig '/home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/sdkconfig'
-- Found Git: /usr/bin/git (found version "2.34.1")
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/mickey/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Check for working C compiler: /home/mickey/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-gcc - skipped
-- Detecting C compile features
-- Detecting C compile features - done
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Check for working CXX compiler: /home/mickey/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32s3-elf-g++ - skipped
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32s3
Processing 5 dependencies:
[1/5] espressif/led_strip (2.5.3)
[2/5] idf (5.2.0)
[3/5] mickey/CANyonero (3ea06cba241cc644103a3fba2b29f9306300f128)
[4/5] mickey/ESPenlaub (0.0.1)
[5/5] mickey/esp-nimble-cpp (1.5.0)
-- Project sdkconfig file /home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/sdkconfig
/home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/sdkconfig:1076 CONFIG_ESP_SLEEP_DEEP_SLEEP_WAKEUP_DELAY was replaced with CONFIG_ESP_SLEEP_WAIT_FLASH_READY_EXTRA_DELAY
-- Compiler supported targets: xtensa-esp-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
[... 122 lines cut ...]
    inlined from 'ble_store_config_delete_rpa_rec' at /home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:780:10,
    inlined from 'ble_store_config_delete' at /home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:922:14:
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:228:9: error: 'memmove' offset [14, 27] is out of the bounds [0, 14] of object 'ble_store_config_rpa_recs' with type 'struct ble_store_value_rpa_rec[1]' [-Werror=array-bounds=]
  228 |         memmove(dst, src, move_count * value_size);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_delete':
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:57:5: note: 'ble_store_config_rpa_recs' declared here
   57 |     ble_store_config_rpa_recs[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~
In function 'ble_store_config_delete_obj',
    inlined from 'ble_store_config_delete_local_irk' at /home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:629:10,
    inlined from 'ble_store_config_delete' at /home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:925:15:
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:228:9: error: 'memmove' offset [23, 45] is out of the bounds [0, 23] of object 'ble_store_config_local_irks' with type 'struct ble_store_value_local_irk[1]' [-Werror=array-bounds=]
  228 |         memmove(dst, src, move_count * value_size);
      |         ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c: In function 'ble_store_config_delete':
/home/mickey/Dokumente/late/misc/esp-idf/components/bt/host/nimble/nimble/nimble/host/store/config/src/ble_store_config.c:62:5: note: 'ble_store_config_local_irks' declared here
   62 |     ble_store_config_local_irks[MYNEWT_VAL(BLE_STORE_MAX_BONDS)];
      |     ^~~~~~~~~~~~~~~~~~~~~~~~~~~
cc1: some warnings being treated as errors
[923/1137] Building C object esp-idf/unity/CMakeFiles/__idf_unity.dir/unity/src/unity.c.obj
ninja: build stopped: subcommand failed.
ninja failed with exit code 1, output of the command is in the /home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/build/log/idf_py_stderr_output_212292 and /home/mickey/Dokumente/late/ecuconnect/firmware/boards/s3devkit/build/log/idf_py_stdout_output_212292
```
```

darshan7patel, 2024-02-12:
Hii @mickeyl ,

I enabled compiler optimization (-02) and compile the blecent / bleprph example for s3 with that flag set to 02 and do not observe any issue (Kindly check the attached log)
So , it may be tied up with environment , or any other setting.

Can you please share the sdkconfig.
[log_comp_opt.txt](https://github.com/espressif/esp-idf/files/14236692/log_comp_opt.txt)

mickeyl, 2024-02-12:
@darshan7patel Yes, I can compile blecent / bleprph with -O2 fine as well. Here's my `sdkconfig`:

[sdkconfig.txt](https://github.com/espressif/esp-idf/files/14236754/sdkconfig.txt)

darshan7patel, 2024-02-13:
Hii @mickeyl ,
Would you kindly apply the patch that is provided and let me know whether it works?
[memmove.txt](https://github.com/espressif/esp-idf/files/14257113/memmove.txt)

mickeyl, 2024-02-13:
@darshan7patel Awesome, that fixes it!

mickeyl, 2024-02-14:
Closing this here, since I have opened it against the proper submodule repository. Thanks again, @darshan7patel -- will keep this patch in my fork for now!

closed 2024-02-14 by mickeyl