espressif/esp-idf #13159: esp_eth_phy_ip101.c:176:5: error: implicit declaration of function 'vTaskDelay'

AtkinEngineering, 2024-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-2020-g4b5b064caf

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### What is the expected behavior?

Compiling with IDF v5.3-dev-1978-g4fb231088d produces no errors.
Compiling with IDF v5.3-dev-2020-g4b5b064caf produces:
esp_eth_phy_ip101.c: In function 'ip101_reset_hw': error: implicit declaration of function 'pdMS_TO_TICKS'

### What is the actual behavior?

v5.3-dev-2020-g4b5b064caf, esp_eth_phy_ip101.c implicit declaration of function 'vTaskDelay' in esp_eth_phy_ip101.c: In function 'ip101_reset_hw'

### Steps to reproduce.

1. Step
2. Step
3. Step
...

### Build or installation Logs.

```plain
C:/Users/dev/esp/master/esp-idf/components/esp_eth/src/esp_eth_phy_ip101.c
C:/Users/dev/esp/master/esp-idf/components/esp_eth/src/esp_eth_phy_ip101.c: In function 'ip101_reset_hw':
C:/Users/dev/esp/master/esp-idf/components/esp_eth/src/esp_eth_phy_ip101.c:176:5: error: implicit declaration of function 'vTaskDelay' [-Werror=implicit-function-declaration]
  176 |     vTaskDelay(pdMS_TO_TICKS(IP101_PHY_POST_RESET_INIT_TIME_MS));
      |     ^~~~~~~~~~
C:/Users/dev/esp/master/esp-idf/components/esp_eth/src/esp_eth_phy_ip101.c:176:16: error: implicit declaration of function 'pdMS_TO_TICKS' [-Werror=implicit-function-declaration]
  176 |     vTaskDelay(pdMS_TO_TICKS(IP101_PHY_POST_RESET_INIT_TIME_MS));
      |                ^~~~~~~~~~~~~
cc1.exe: some warnings being treated as errors
```

AtkinEngineering, 2024-02-12:
Please close. Once enabling SPI Ethernet, the error has disappeared.

closed 2024-02-12 by AtkinEngineering

kostaond, 2024-03-22:
Fixed by eb8985886d492603a4d745fba1426c44ee2bc3eb