espressif/esp-idf #13160: Can ESP-S host mode be used to read data from serial device ?

khayamgondal, 2024-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I have a 3D printer that exposes serial over USB. Can I use this library to read that data if printer is connected to an ESP S3's host mode port?

tore-espressif, 2024-02-13:
Hello @khayamgondal , yes, we have support for USB Host serial.

Here are some examples, that can help you started: https://github.com/espressif/esp-idf/tree/master/examples/peripherals/usb/host/cdc

But the chances are that you will need to implement a device driver specific for your device.

Could you please share USB descriptors of you 3D printer? You can find them by connecting the printer to PC and then `lsusb -v` on linux, or using [USB Device Tree Viewer](https://www.uwe-sieber.de/usbtreeview_e.html) on windows

tore-espressif, 2024-03-05:
Closing this for now due to lack of feedback. Feel free to reopen