espressif/esp-idf #13162: ESP32-C6

thauti, 2024-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5.2.0

### Espressif SoC revision.

ESP32-C6

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

ESP32-C6-DEVKIT-1

### Power Supply used.

USB

### What is the expected behavior?

I expected it to write some data to UART stdout without the need of end my strings with "\n" (newline)

### What is the actual behavior?

Instead it requires to \n to get the output working.
I need to output some binary data to stdout and "\n" (0x10) add unwanted values in my data stream

### Steps to reproduce.

With :
```
printf("Hello world \n");
```
It works

With : 
```
printf("Hello world ");
```
It doesn't display anything, until I print '\n' 

### More Information.

I tried to flush stdout with fflush and/or disable the buffer mecanism with setvbuf(stdout, NULL, _IONBF, 0); , but it changes nothing

igrr, 2024-02-11:
> I tried to flush stdout with fflush and/or disable the buffer mecanism with setvbuf(stdout, NULL, _IONBF, 0);

There are typically two levels of buffering: in stdio (C standard library) and at the underlying device driver level. You got it right that the stdio buffer can be flushed using `fflush` or disabled using `setvbuf`. In addition to that, you have to flush the lower-level buffer using `fsync(fileno(stdout));`

([Docs](https://pubs.opengroup.org/onlinepubs/009695399/functions/fsync.html))

thauti, 2024-02-11:
> There are typically two levels of buffering: in stdio (C standard library) and at the underlying device driver level. You got it right that the stdio buffer can be flushed using `fflush` or disabled [...]

Thanks for you help
But I still have the problem using :
```
        printf("Hello World 2");
        fsync(fileno(stdout));        
        fflush(stdout);
```

igrr, 2024-02-11:
It should be the other way around:

```c++
        printf("Hello World 2");        
        fflush(stdout);
        fsync(fileno(stdout));
```

because fflush will transfer the `stdout` file buffer to the underlying device, and `fsync` will flush the device buffer.

thauti, 2024-02-12:
> It should be the other way around:
> 
> ```c++
>         printf("Hello World 2");        
>         fflush(stdout);
>         fsync(fileno(stdout));
> ```
> 
> because fflush will transfer the `stdout` file buffer to the underlying device, and `fsync` will flush the device buffer.

I still have the problem ... :/

igrr, 2024-02-13:
Can you please describe how you are connecting your board (via the usb-serial or the built-in USB) and attach your sdkconfig file? 
If you add the newline, do you get the output?

thauti, 2024-02-13:
I'm using this one
[image]

With VScode in Uart mode 
[image]

Sdkconfig
```
#
# Automatically generated file. DO NOT EDIT.
# Espressif IoT Development Framework (ESP-IDF) 5.2.0 Project Configuration
#
CONFIG_SOC_ADC_SUPPORTED=y
CONFIG_SOC_DEDICATED_GPIO_SUPPORTED=y
CONFIG_SOC_UART_SUPPORTED=y
CONFIG_SOC_GDMA_SUPPORTED=y
CONFIG_SOC_AHB_GDMA_SUPPORTED=y
CONFIG_SOC_GPTIMER_SUPPORTED=y
CONFIG_SOC_PCNT_SUPPORTED=y
CONFIG_SOC_MCPWM_SUPPORTED=y
CONFIG_SOC_TWAI_SUPPORTED=y
CONFIG_SOC_ETM_SUPPORTED=y
CONFIG_SOC_PARLIO_SUPPORTED=y
CONFIG_SOC_BT_SUPPORTED=y
CONFIG_SOC_IEEE802154_SUPPORTED=y
CONFIG_SOC_ASYNC_MEMCPY_SUPPORTED=y
CONFIG_SOC_USB_SERIAL_JTAG_SUPPORTED=y
CONFIG_SOC_TEMP_SENSOR_SUPPORTED=y
CONFIG_SOC_WIFI_SUPPORTED=y
CONFIG_SOC_SUPPORTS_SECURE_DL_MODE=y
CONFIG_SOC_ULP_SUPPORTED=y
CONFIG_SOC_LP_CORE_SUPPORTED=y
CONFIG_SOC_EFUSE_KEY_PURPOSE_FIELD=y
CONFIG_SOC_EFUSE_SUPPORTED=y
CONFIG_SOC_RTC_FAST_MEM_SUPPORTED=y
CONFIG_SOC_RTC_MEM_SUPPORTED=y
CONFIG_SOC_I2S_SUPPORTED=y
CONFIG_SOC_RMT_SUPPORTED=y
CONFIG_SOC_SDM_SUPPORTED=y
CONFIG_SOC_GPSPI_SUPPORTED=y
CONFIG_SOC_LEDC_SUPPORTED=y
CONFIG_SOC_I2C_SUPPORTED=y
CONFIG_SOC_SYSTIMER_SUPPORTED=y
CONFIG_SOC_SUPPORT_COEXISTENCE=y
CONFIG_SOC_AES_SUPPORTED=y
CONFIG_SOC_MPI_SUPPORTED=y
CONFIG_SOC_SHA_SUPPORTED=y
CONFIG_SOC_HMAC_SUPPORTED=y
CONFIG_SOC_DIG_SIGN_SUPPORTED=y
CONFIG_SOC_ECC_SUPPORTED=y
CONFIG_SOC_FLASH_ENC_SUPPORTED=y
CONFIG_SOC_SECURE_BOOT_SUPPORTED=y
CONFIG_SOC_SDIO_SLAVE_SUPPORTED=y
CONFIG_SOC_BOD_SUPPORTED=y
CONFIG_SOC_APM_SUPPORTED=y
CONFIG_SOC_PMU_SUPPORTED=y
CONFIG_SOC_PAU_SUPPORTED=y
CONFIG_SOC_LP_TIMER_SUPPORTED=y
CONFIG_SOC_LP_AON_SUPPORTED=y
CONFIG_SOC_LP_PERIPHERALS_SUPPORTED=y
CONFIG_SOC_LP_I2C_SUPPORTED=y
CONFIG_SOC_ULP_LP_UART_SUPPORTED=y
CONFIG_SOC_CLK_TREE_SUPPORTED=y
CONFIG_SOC_ASSIST_DEBUG_SUPPORTED=y
CONFIG_SOC_WDT_SUPPORTED=y
CONFIG_SOC_SPI_FLASH_SUPPORTED=y
CONFIG_SOC_XTAL_SUPPORT_40M=y
CONFIG_SOC_AES_SUPPORT_DMA=y
CONFIG_SOC_AES_GDMA=y
CONFIG_SOC_AES_SUPPORT_AES_128=y
[... 1888 lines cut ...]
CONFIG_TCP_WND_DEFAULT=5760
CONFIG_TCP_RECVMBOX_SIZE=6
CONFIG_TCP_QUEUE_OOSEQ=y
CONFIG_TCP_OVERSIZE_MSS=y
# CONFIG_TCP_OVERSIZE_QUARTER_MSS is not set
# CONFIG_TCP_OVERSIZE_DISABLE is not set
CONFIG_UDP_RECVMBOX_SIZE=6
CONFIG_TCPIP_TASK_STACK_SIZE=3072
CONFIG_TCPIP_TASK_AFFINITY_NO_AFFINITY=y
# CONFIG_TCPIP_TASK_AFFINITY_CPU0 is not set
CONFIG_TCPIP_TASK_AFFINITY=0x7FFFFFFF
# CONFIG_PPP_SUPPORT is not set
CONFIG_ESP32_PTHREAD_TASK_PRIO_DEFAULT=5
CONFIG_ESP32_PTHREAD_TASK_STACK_SIZE_DEFAULT=3072
CONFIG_ESP32_PTHREAD_STACK_MIN=768
CONFIG_ESP32_PTHREAD_TASK_CORE_DEFAULT=-1
CONFIG_ESP32_PTHREAD_TASK_NAME_DEFAULT="pthread"
CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ABORTS=y
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_FAILS is not set
# CONFIG_SPI_FLASH_WRITING_DANGEROUS_REGIONS_ALLOWED is not set
CONFIG_SUPPRESS_SELECT_DEBUG_OUTPUT=y
CONFIG_SUPPORT_TERMIOS=y
CONFIG_SEMIHOSTFS_MAX_MOUNT_POINTS=1
# End of deprecated options
```

igrr, 2024-02-13:
Okay, I understand the issue now. This is a bug in IDF indeed. You are using not the UART, but the USB_SERIAL_JTAG peripheral (since the cable is plugged into the built-in USB port of the C6). However when USB_SERIAL_JTAG as used as secondary output (CONFIG_ESP_CONSOLE_SECONDARY_USB_SERIAL_JTAG=y), we don't propagate fsync calls to it over here:
https://github.com/espressif/esp-idf/blob/4b5b064caf951a48b48a39293d625b5de2132719/components/vfs/vfs_console.c#L110

You currently have to options:

1. Plug the cable into the other USB port, which is connected to the UART via a USB-serial bridge.
2. Or, change the primary channel of console output to USB_SERIAL_JTAG (enable CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG).

We'll fix the issue, but it might take some time since part of the team is currently away for the Chinese New Year holidays.

thauti, 2024-02-13:
Thank you 👍 
When I change the port, I got weird output, but the 2. did the trick (in Putty, VSCode doesn't seems to show things that are not ESP_LOG*)

closed 2024-04-17 by espressif-bot