espressif/esp-idf #13163: ESP-Modbus不使用定时器进行3.5T符号时间时第一个数据包没有回复

Zhicon, 2024-02-11:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

V5.1

### Espressif SoC revision.

ESP32-S2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

Custom Board

### Power Supply used.

USB

### What is the expected behavior?

失能Modbus stack use timer for 3.5T symbol time measurement后，modbus功能应该不受影响

### What is the actual behavior?

失能Modbus stack use timer for 3.5T symbol time measurement后，esp32-s2上电后主机发送的第一个modbus数据包没有回复

### Steps to reproduce.

失能Modbus stack use timer for 3.5T symbol time measurement后就会出现这个问题

### More Information.

最终定位到mbrtu.c文件中的 void eMBRTUStart( void )函数
`void eMBRTUStart( void )
{
    ENTER_CRITICAL_SECTION(  );
    /* Initially the receiver is in the state STATE_RX_INIT. we start
     * the timer and if no character is received within t3.5 we change
     * to STATE_RX_IDLE. This makes sure that we delay startup of the
     * modbus protocol stack until the bus is free.
     */
    eRcvState = STATE_RX_INIT;
    vMBPortSerialEnable( TRUE, FALSE );
    vMBPortTimersEnable(  );

    EXIT_CRITICAL_SECTION(  );
}`
将其修改为
`void eMBRTUStart( void )
{
    ENTER_CRITICAL_SECTION(  );
    /* Initially the receiver is in the state STATE_RX_INIT. we start
     * the timer and if no character is received within t3.5 we change
     * to STATE_RX_IDLE. This makes sure that we delay startup of the
     * modbus protocol stack until the bus is free.
     */
    eRcvState = STATE_RX_INIT;
    vMBPortSerialEnable( TRUE, FALSE );
#if CONFIG_FMB_TIMER_PORT_ENABLED
    vMBPortTimersEnable(  );
#else
    pxMBPortCBTimerExpired();
#endif
    EXIT_CRITICAL_SECTION(  );
}`
我只测试了modbus rtu功能，但我想在ascii模式下应该也有这个问题

alisitsyn, 2024-02-19:
@Zhicon , 

Thank you for the issue. 
The Modbus RTU starts from `STATE_RX_IDLE` stage and it skips the incoming data received before the 3.5T.

> 失能Modbus stack use timer for 3.5T symbol time measurement后，modbus功能应该不受影响
> 失能Modbus stack use timer for 3.5T symbol time measurement后，esp32-s2上电后主机发送的第一个modbus数据包没有回复

When the `CONFIG_FMB_TIMER_PORT_ENABLED` is not enabled the stack ignores the first incoming packet. This behavior is confirmed as incorrect.
Your proposed code can be used to handle the Init state correctly.

> 我只测试了modbus rtu功能，但我想在ascii模式下应该也有这个问题

The ASCII mode has different logic and is able to start from IDLE state and can process the first packet correctly.

The fix is planned to be implemented in next release.

alisitsyn, 2024-04-29:
The issue solved with commit [8165909be4328fd6ce14e94dbd4932bf2a0ec4a9](https://gitlab.espressif.cn:6688/idf/esp-modbus/-/commit/8165909be4328fd6ce14e94dbd4932bf2a0ec4a9).

closed 2024-04-29 by alisitsyn