espressif/esp-idf #13167: Bluetooth AVRCP: linking fails with "undefined reference to `esp_avrc_ct_init'"

eni23, 2024-02-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-D0WD-V3

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32-lyrat-mini v1.2

### Power Supply used.

USB

### What is the expected behavior?

Compilation & linking suceeds

### What is the actual behavior?

linking fails with:

```
.../main/main.c:180: undefined reference to `esp_avrc_ct_init'
collect2: error: ld returned 1 exit status
```

### Steps to reproduce.

On latest stable IDF (`v5.1.2`), i cannot use the BT AVRCP functions. 

My `CMakeLists.txt` has `REQUIRE bt` inserted, but when i try to use it:

```
[...]
#include "esp_avrc_api.h"

[...]
    esp_err_t avrpc_err;
    avrpc_err = esp_avrc_ct_init();
[...]
```

It compiles correctly, but the linking fails:

```
.../main/main.c:180: undefined reference to `esp_avrc_ct_init'
collect2: error: ld returned 1 exit status
```

any ideas why?

eni23, 2024-02-12:
Additional info: the linking also fails with `esp-idf` on `master`  branch

esp-qing, 2024-02-19:
Hi, can you show me the contents of the `sdkconfig` ?

eni23, 2024-02-19:
> Hi, can you show me the contents of the `sdkconfig` ?

Sure (i hope the `sdkconfig.defaults`) is enough:

```
CONFIG_BT_ENABLED=y
CONFIG_BT_BLE_ENABLED=n
CONFIG_BTDM_CTRL_MODE_BR_EDR_ONLY=y
CONFIG_BTDM_CTRL_BR_EDR_MAX_SYNC_CONN=1
CONFIG_BT_BLUEDROID_ENABLED=y
CONFIG_BT_CLASSIC_ENABLED=y
CONFIG_BT_HFP_ENABLE=y
CONFIG_BT_HFP_CLIENT_ENABLE=y
```

esp-qing, 2024-02-20:
Hi, please add `CONFIG_BT_A2DP_ENABLE=y` in `sdkconfig.defaults`

esp-qing, 2024-02-23:
Hi, @eni23 
In esp-idf, enabling both AVRCP and A2DP is required simultaneously. It is currently not possible to use AVRCP independently, as the initialization of AVRCP is integrated within the A2DP initialization function. It is possible to use HFP together with A2DP in esp-idf and would this meet your needs?

closed 2025-04-03 by Alvin1Zhang