espressif/esp-idf #13168: Is EHOSTUNREACH 113 or 118 ?

dannybackx, 2024-02-12:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2 rc1

### Espressif SoC revision.

esp32

### Operating System used.

Linux

### How did you build your project?

Command line with idf.py

### If you are using Windows, please specify command line type.

None

### Development Kit.

esp32 d1 mini

### Power Supply used.

USB

### What is the expected behavior?

My code gets 113 as error code when trying to connect to a node that's currently powered off, so that's normal.
When I tried filtering that condition and writing this as
```
if (error == EHOSTUNREACH)
```
this didn't work.

Turns out EHOSTUNREACH is 113 in $IDF_PATH/components/lwip/lwip/src/include/lwip/errno.h but 118 in $HOME/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/xtensa-esp32-elf/include/sys/errno.h .

I'm guessing something is wrong.

The 113 number returned (by getsockname) could also mean ECONNABORTED I guess, like it's in the other include file.

### What is the actual behavior?

113 but what does it mean

### Steps to reproduce.

Write a socket program that connects to a node currently powered off.

igrr, 2024-02-18:
That header file is the standard part of lwip, and the values are guarded by `#ifdef LWIP_PROVIDE_ERRNO`:

https://github.com/espressif/esp-lwip/blob/a45be9e438f6cf9c54ec150581819c3b95d5af6b/src/include/lwip/errno.h#L46

I think your editor or an IDE should not index these definitions, so there shouldn't be much chance for a confusion. On the other hand, removing those definitions would make our fork more divergent from upstream LWIP.

> how can anyone be sure of that

One possible solution would be to add errno code reference to the programming guide, similar to esp_err_t reference we have already.

david-cermak, 2024-03-11:
Closing as we already have a task to add the `errno` codes to the docs.
Please subscribe to https://github.com/espressif/esp-idf/issues/8546 to get notifications on progress.

closed 2024-03-11 by david-cermak

jcecconi, 2024-08-19:
Hi, has anybody solved this issue? I have the same problem... When I try to connect to an unreachable device with ESP-IDF v5.2, the connect function returns -1, with errno set to 113 (which equals ECONNABORTED), while EHOSTUNREACH is equal to 118."

Let me know if you need any further adjustments!