espressif/esp-idf #13172: SPI peripheral / bus numbers on ESP32S3 wrong and overly limited

anm, 2024-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

The ESP32S3 datasheet says it has four SPI controllers, numbered 0, 1, 2, and 3.

`esp32-hal-spi.c` says:
`#define SPI_COUNT           (2)`

This causes it to reject starting the peripherals numbered 2 and 3.

`esp32-hal-spi.h` also defines only:
```
#define FSPI  0
#define HSPI  1
```

atanisoft, 2024-02-13:
Those filenames look like they are from arduino-esp32 and not esp-idf.

Note that even though there are more SPI controllers, some are used internally and are not available for user / application code.

anm, 2024-02-13:
So it is, I've moved report to that repo.

closed 2024-02-13 by anm