espressif/esp-idf #13173: Disabling IPV4 casing compilation error

ShubhadaHarkut-eaton, 2024-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

4.4

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

IPV6 working standalone with ipv4 support completly disabled

### What is the actual behavior?

Disabling IVP4 and building the idf ethernet example code only with IPV6 support enabled leading to many compilation error
idf version used : 4.4
os : windows
ESP : ESP32-WROVER-B

disabling LWIP_IPV4 in opts.h and enabling LWIP_IPV6 in the same file along with other menuconfig configurations for rnabling IPV6

### Steps to reproduce.

disabling LWIP_IPV4 in opts.h and enabling LWIP_IPV6 in the same file along with other menuconfig configurations for rnabling IPV6

### Build or installation Logs.

```plain
Building C object esp-idf/esp_netif/CMakeFiles/__idf_esp_netif.dir/lwip/esp_netif_lwip_defaults.c.obj
FAILED: esp-idf/esp_netif/CMakeFiles/__idf_esp_netif.dir/lwip/esp_netif_lwip_defaults.c.obj
C:\Espressif\tools\xtensa-esp32-elf\esp-2021r2-patch2-8.4.0\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe  -Iconfig -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/lwip -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/private_include -IC:/Espressif/frameworks/esp-idf-v4.4/components/newlib/platform_include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions/freertos -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/port/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/freertos/include/esp_additions -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/include/soc/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_hw_support/port/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/heap/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/log/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/sntp -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/lwip/src/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/lwip/port/esp32/include/arch -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/. -IC:/Espressif/frameworks/esp-idf-v4.4/components/soc/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/hal/platform_port/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/include/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_rom/esp32 -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_common/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/soc -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_system/port/public_compat -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/xtensa/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/driver/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_pm/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ringbuf/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/efuse/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/vfs/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_wifi/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_event/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_eth/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/tcpip_adapter/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_phy/esp32/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_ipc/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/app_trace/include -IC:/Espressif/frameworks/esp-idf-v4.4/components/esp_timer/include -mlongcalls -Wno-frame-address -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -ggdb -Og -fmacro-prefix-map=C:/BLR/platform/edge-rtos-app-esp32/esp32_common/esp-idf/examples/protocols/http2_request=. -fmacro-prefix-map=C:/Espressif/frameworks/esp-idf-v4.4=IDF -fstrict-volatile-bitfields -Wno-error=unused-but-set-variable -fno-jump-tables -fno-tree-switch-conversion -std=gnu99 -Wno-old-style-declaration -D_GNU_SOURCE -DIDF_VER=\"v4.4\" -DESP_PLATFORM -D_POSIX_READER_WRITER_LOCKS -MD -MT esp-idf/esp_netif/CMakeFiles/__idf_esp_netif.dir/lwip/esp_netif_lwip_defaults.c.obj -MF esp-idf\esp_netif\CMakeFiles\__idf_esp_netif.dir\lwip\esp_netif_lwip_defaults.c.obj.d -o esp-idf/esp_netif/CMakeFiles/__idf_esp_netif.dir/lwip/esp_netif_lwip_defaults.c.obj -c C:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/lwip/esp_netif_lwip_defaults.c
In file included from C:/Espressif/frameworks/esp-idf-v4.4/components/tcpip_adapter/include/tcpip_adapter_types.h:19,
                 from C:/Espressif/frameworks/esp-idf-v4.4/components/tcpip_adapter/include/tcpip_adapter.h:24,
                 from C:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/include/esp_netif.h:35,
                 from C:/Espressif/frameworks/esp-idf-v4.4/components/esp_netif/lwip/esp_netif_lwip_defaults.c:15:
C:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/dhcpserver/dhcpserver.h:45:2: error: unknown type name 'ip4_addr_t'
  ip4_addr_t start_ip;
  ^~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/dhcpserver/dhcpserver.h:46:2: error: unknown type name 'ip4_addr_t'
  ip4_addr_t end_ip;
  ^~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/dhcpserver/dhcpserver.h:62:2: error: unknown type name 'ip4_addr_t'
  ip4_addr_t ip;
  ^~~~~~~~~~
C:/Espressif/frameworks/esp-idf-v4.4/components/lwip/include/apps/dhcpserver/dhcpserver.h:89:39: error: unknown type name 'ip4_addr_t'; did you mean 'ip_addr_t'?
 void dhcps_start(struct netif *netif, ip4_addr_t ip);
                                       ^~~~~~~~~~
```

sgryphon, 2024-02-24:
I am not sure if it is relevant for your scenario, but I recently put up a pull request to update the protocol examples to work with all/any network configuration. The branch is here: https://github.com/espressif/esp-idf/pull/13250

The update examples work out of the box (no config or code changes) with all network types: IPv4-only, IPv6-only, and dual stack. (You do need to provide config for things like your WiFi password)

Note that this is not the same as turning off IPv4 -- IPv4 is still turned on, just not used. The updated example works in an IPv6-only network, if that is your objective. See the example outputs for IPv6-only network. If you have DNS64/NAT64 available, then it can even connect from an IPv6-only network to an IPv4-only server.

However, IPv4 is still actually on.

If you know IPv4 is not necessary, e.g. if you are developing a Matter device where only IPv6 is required, and want to reduce memory size and development/testing effort, then you could also turn off IPv4 completely.

sgryphon, 2024-02-24:
I just tried turning off LWIP_IPV4 for the updated example, and it does break compilation (in the PPP component). The example still works on an IPv6-only network, it just needs IPv4 turned on (although not being used).

I will see if there is a quick fix for the example.

espressif-abhikroy, 2024-02-28:
The option to disable IPv4 is not available for IDF release v4.4 and v5.0.
It's available from IDF release v5.1 onwards.
Closing the other issue as duplicate
https://github.com/espressif/esp-lwip/issues/64

espressif-abhikroy, 2024-02-28:
The option to disable IPv4 is not available for IDF release v4.4 and v5.0.
It's available from IDF release v5.1 onwards.
Closing the issue as it's duplicate of 
https://github.com/espressif/esp-lwip/issues/64

closed 2024-02-28 by espressif-abhikroy