espressif/esp-idf #13175: gpio_dump_all_io_configuration() missing

nicklasb, 2024-02-13:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

5

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

I expected gpio_dump_all_io_configuration to exist, as prominently mentioned by the documentation at https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/peripherals/gpio.html.

### What is the actual behavior?

But oddly it seems nowhere to be found. Nowhere in the source trees or in any header files, only in the documentation.
Doesn't seem to be ifdef:ed or anything, just plain missing. Even from the Espressif repos. Or in any repo.

### Steps to reproduce.

Try to use gpio_dump_all_io_configuration() in any situation.

### Build or installation Logs.

```plain
Not applicable.
```

AxelLin, 2024-02-15:
gpio_dump_io_configuration() only available in IDF-5.1+.
It does not exist in 5.0 branch, see https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/gpio.html

nicklasb, 2024-02-15:
Ok. I saw it in the master branch of documentation but couldn't find in the master branch of the source. But good, then I know.

closed 2024-02-15 by nicklasb

Alvin1Zhang, 2024-02-19:
@AxelLin Thanks for sharing the link, @nicklasb good to hear the issue is resolved.

baldhead69, 2024-09-25:
> gpio_dump_io_configuration() only available in IDF-5.1+. It does not exist in 5.0 branch, see https://docs.espressif.com/projects/esp-idf/en/v5.0.5/esp32/api-reference/peripherals/gpio.html

I also tried to use this function on esp32-s3 and i couldn't.
I didn't find it in esp-idf v5.3.1.

https://esp32.com/viewtopic.php?f=13&t=41942&p=138055#p138055

OnceWas, 2024-10-25:
The function is not present in 5.4 either:

```
> idf.py --version
ESP-IDF v5.4-dev-3602-ga97a7b0962
```

`error: implicit declaration of function 'gpio_dump_all_io_configuration'; did you mean 'gpio_dump_io_configuration'?`

baldhead69, 2025-02-19:
This question was closed why ?

Was the problem solved ?

nicklasb, 2025-02-19:
> This question was closed why ?
> 
> Was the problem solved ?

From my perspective it was, I got the answer that it was in 5.1, and I think I found it there somewhere. But if it is missing again, perhaps it need reopening? @AxelLin ?