espressif/esp-idf #13183: esp_ble_gattc_write_char() and esp_ble_gattc_read_char() are not working when called in main

Manster, 2024-02-15:
esp_ble_gattc_write_char() and esp_ble_gattc_read_char()  i want to call them from main for my functionality sake but its not working as expected it successfully goes to the Write and Read events showing read write success but its not writing nor reading the handles are correct i checked twice so what can be the issue i have attached client code as well as server code and Output of both.
[CLIENT_CODE.txt](https://github.com/espressif/esp-idf/files/14291490/CLIENT_CODE.txt)
[SERVER_CODE.txt](https://github.com/espressif/esp-idf/files/14291494/SERVER_CODE.txt)

@Weijian-Espressif 
@chegewara 

Client Output:
I (18837) SEC_GATTC_DEMO: ESP_GAP_BLE_SCAN_RESULT_EVT

I (18837) SEC_GATTC_DEMO: searched device ESP_BLE_SECURITY

I (18857) SEC_GATTC_DEMO: Stop scan successfully
I (19177) SEC_GATTC_DEMO: EVT 40, gattc if 1
I (19177) SEC_GATTC_DEMO: EVT 2, gattc if 1
I (19177) SEC_GATTC_DEMO: open success
I (19177) SEC_GATTC_DEMO: REMOTE BDA:
I (19187) SEC_GATTC_DEMO: 43 75 af 2d 06 4a 
I (19337) SEC_GATTC_DEMO: remote BD_ADDR: 4375af2d064a
I (19347) SEC_GATTC_DEMO: address type = 1
I (19347) SEC_GATTC_DEMO: pair status = success
I (19347) SEC_GATTC_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (19987) SEC_GATTC_DEMO: EVT 46, gattc if 1
I (20037) SEC_GATTC_DEMO: EVT 18, gattc if 1
I (20037) SEC_GATTC_DEMO: ESP_GATTC_CFG_MTU_EVT, Status 0, MTU 200, conn_id 0
I (20047) SEC_GATTC_DEMO: EVT 7, gattc if 1
I (20047) SEC_GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (20057) SEC_GATTC_DEMO: start handle 40 end handle 47 current handle value 40
I (20067) SEC_GATTC_DEMO: UUID16: ff
I (20067) SEC_GATTC_DEMO: EVT 6, gattc if 1
I (20067) SEC_GATTC_DEMO: Get service information from remote device
I (20077) SEC_GATTC_DEMO: EVT 7, gattc if 1
I (20087) SEC_GATTC_DEMO: SEARCH RES: conn_id = 0 is primary service 1
I (20087) SEC_GATTC_DEMO: start handle 48 end handle 65535 current handle value 48
I (20097) SEC_GATTC_DEMO: UUID16: ee
I (20107) SEC_GATTC_DEMO: EVT 6, gattc if 1
I (20107) SEC_GATTC_DEMO: Get service information from remote device
I (21007) SEC_GATTC_DEMO: EVT 4, gattc if 1
HiiiI (21007) SEC_GATTC_DEMO: Write char success 
I (21007) SEC_GATTC_DEMO: EVT 3, gattc if 1
HiI (21007) SEC_GATTC_DEMO: Read char success 
I (21017) SEC_GATTC_DEMO: EVT 4, gattc if 1
HiiiI (21017) SEC_GATTC_DEMO: Write char success 
I (21027) main_task: Returned from app_main()
I (21057) SEC_GATTC_DEMO: EVT 3, gattc if 1
HiI (21057) SEC_GATTC_DEMO: Read char success 

Server Output:
I (566) main_task: Calling app_main()
I (576) BTDM_INIT: BT controller compile version [ec4ac65]
I (586) BTDM_INIT: Bluetooth MAC: 78:21:84:cf:22:b6
I (586) phy_init: phy_version 4780,16b31a7,Sep 22 2023,20:42:16
I (906) SEC_GATTS_DEMO: app_main init bluetooth
I (1006) SEC_GATTS_DEMO: The number handle = 8
I (1016) main_task: Returned from app_main()
I (1036) SEC_GATTS_DEMO: The number handle = 8
I (1036) SEC_GATTS_DEMO: advertising start success
I (6516) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
I (6516) SEC_GATTS_DEMO: ESP_GATTS_CONNECT_EVT
E (6516) BT_APPL: earlier enc was not done for same device

I (6666) SEC_GATTS_DEMO: remote BD_ADDR: 687c53b2b7ab
I (6666) SEC_GATTS_DEMO: address type = 1
I (6666) SEC_GATTS_DEMO: pair status = success
I (6666) SEC_GATTS_DEMO: auth mode = ESP_LE_AUTH_REQ_SC_BOND
I (6676) SEC_GATTS_DEMO: Bonded devices number : 1

I (6686) SEC_GATTS_DEMO: Bonded devices list : 1

I (6686) SEC_GATTS_DEMO: 68 7c 53 b2 b7 ab

zhp0406, 2024-02-20:
@Manster 
Can you provide me with the complete server-side code? I can't compile it on my end.

```
../main/example_ble_sec_gatts_demo.c:43:30: error: 'IDX_NB' undeclared here (not in a function); did you mean 'HRS_IDX_NB'?
   43 | static uint16_t handle_table[IDX_NB];
      |                              ^~~~~~
      |                              HRS_IDX_NB
../main/example_ble_sec_gatts_demo.c:44:31: error: 'IDX2_NB' undeclared here (not in a function)
   44 | static uint16_t handle2_table[IDX2_NB];
      |                               ^~~~~~~
../main/example_ble_sec_gatts_demo.c:173:6: error: 'SVC1' undeclared here (not in a function)
  173 |     [SVC1]                    =
      |      ^~~~
../main/example_ble_sec_gatts_demo.c:178:6: error: 'CHAR1_DEC' undeclared here (not in a function)
  178 |     [CHAR1_DEC]            =
      |      ^~~~~~~~~
../main/example_ble_sec_gatts_demo.c:183:6: error: 'CHAR1_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  183 |     [CHAR1_VAL]             =
      |      ^~~~~~~~~
      |      CHAR_MAX
../main/example_ble_sec_gatts_demo.c:188:6: error: 'CHAR1_CCD' undeclared here (not in a function)
  188 |     [CHAR1_CCD]        =
      |      ^~~~~~~~~
../main/example_ble_sec_gatts_demo.c:193:6: error: 'CHAR2_DEC' undeclared here (not in a function)
  193 |     [CHAR2_DEC]  =
      |      ^~~~~~~~~
../main/example_ble_sec_gatts_demo.c:198:6: error: 'CHAR2_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  198 |     [CHAR2_VAL]   =
      |      ^~~~~~~~~
      |      CHAR_MAX
../main/example_ble_sec_gatts_demo.c:203:6: error: 'CHAR3_DEC' undeclared here (not in a function)
  203 |     [CHAR3_DEC]          =
      |      ^~~~~~~~~
../main/example_ble_sec_gatts_demo.c:208:6: error: 'CHAR3_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  208 |     [CHAR3_VAL]             =
      |      ^~~~~~~~~
      |      CHAR_MAX
../main/example_ble_sec_gatts_demo.c:216:6: error: 'SVC2' undeclared here (not in a function)
  216 |     [SVC2]                    =
      |      ^~~~
../main/example_ble_sec_gatts_demo.c:221:6: error: 'CHAR21_DEC' undeclared here (not in a function)
  221 |     [CHAR21_DEC]            =
      |      ^~~~~~~~~~
../main/example_ble_sec_gatts_demo.c:226:6: error: 'CHAR21_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  226 |     [CHAR21_VAL]             =
      |      ^~~~~~~~~~
      |      CHAR_MAX
../main/example_ble_sec_gatts_demo.c:231:6: error: 'CHAR21_CCD' undeclared here (not in a function)
  231 |     [CHAR21_CCD]        =
      |      ^~~~~~~~~~
../main/example_ble_sec_gatts_demo.c:236:6: error: 'CHAR22_DEC' undeclared here (not in a function)
  236 |     [CHAR22_DEC]  =
      |      ^~~~~~~~~~
../main/example_ble_sec_gatts_demo.c:241:6: error: 'CHAR22_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  241 |     [CHAR22_VAL]   =
      |      ^~~~~~~~~~
      |      CHAR_MAX
../main/example_ble_sec_gatts_demo.c:246:6: error: 'CHAR23_DEC' undeclared here (not in a function)
  246 |     [CHAR23_DEC]          =
      |      ^~~~~~~~~~
../main/example_ble_sec_gatts_demo.c:251:6: error: 'CHAR23_VAL' undeclared here (not in a function); did you mean 'CHAR_MAX'?
  251 |     [CHAR23_VAL]             =
```

zhp0406, 2024-02-28:
Has your issue been resolved?

closed 2024-03-10 by Alvin1Zhang