espressif/esp-idf #13184: transport_base: tcp_write error, errno = No more processes, transport_ws: Error write header

tao-a-man, 2024-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-S3

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

PowerShell

### Development Kit.

esp-idf-v5.1.2

### Power Supply used.

USB

### What is the expected behavior?

I want create 4 Task 
[image]

### What is the actual behavior?

When i call API in task `Task_READER` so much times I get error
[image]

### Steps to reproduce.

1.Build
2.Call API so much times 

### Debug Logs.

```plain
transport_base: tcp_write error, errno = No more processes
transport_ws: Error write header
```

### More Information.

code in Task_WS (Websocket)
```
xTaskCreate([](void *pvParameters)
                {
                    esp_websocket_client_config_t config = {
                        .uri = "ws://10.17.215.214:7911",
                        // end cofig
                    };

                    client = esp_websocket_client_init(&config);
                    esp_websocket_register_events(
                        client, WEBSOCKET_EVENT_ANY, ([](void *handler_args, esp_event_base_t base, int32_t event_id, void *event_data)
                                                      {
                                                          esp_websocket_event_data_t *data = (esp_websocket_event_data_t *)event_data;
                                                          switch (event_id)
                                                          {
                                                          case WEBSOCKET_EVENT_CONNECTED:
                                                              ESP_LOGI(TAG, "WEBSOCKET_EVENT_CONNECTED");
                                                              //     esp_websocket_client_send(client, data, len, portMAX_DELAY);

                                                              slint::invoke_from_event_loop([&, ui = slint::ComponentWeakHandle(ui)]
                                                                                            {
                                                                                                auto ui_lock = *ui.lock();
                                                                                                ui_lock->set_IsConnectWS(true);

                                                                                                // end event
                                                                                            });
                                                              break;
                                                          case WEBSOCKET_EVENT_DISCONNECTED:
                                                              ESP_LOGI(TAG, "WEBSOCKET_EVENT_DISCONNECTED");

                                                              slint::invoke_from_event_loop([&, ui = slint::ComponentWeakHandle(ui)]
                                                                                            {
                                                                                                auto ui_lock = *ui.lock();
                                                                                                ui_lock->set_IsConnectWS(false);
                                                                                                // end event
                                                                                            });
                                                              break;
                                                          case WEBSOCKET_EVENT_DATA:
                                                              ESP_LOGI(TAG, "WEBSOCKET_EVENT_DATA");
                                                              ESP_LOGI(TAG, "Received opcode=%d", data->op_code);
                                                              ESP_LOGW(TAG, "Received=%.*s", data->data_len, (char *)data->data_ptr);
                                                              ESP_LOGW(TAG, "Total payload length=%d, data_len=%d, current payload offset=%d\r\n", data->payload_len, data->data_len, data->payload_offset);

                                                              break;
                                                          case WEBSOCKET_EVENT_ERROR:
                                                              ESP_LOGI(TAG, "WEBSOCKET_EVENT_ERROR");
                                                              break;
                                                          }
                                                          // end event
                                                      }),
                        (void *)client);

                    esp_websocket_client_start(client);
                    while (1)
                    {

                        vTaskDelay(2000 / portTICK_PERIOD_MS);
                    }
                    esp_websocket_client_stop(client);
                    ESP_LOGI(TAG, "Websocket Stopped");
                    esp_websocket_client_destroy(client);
                    vTaskDelete(NULL);
                    // end task
                },
                "Task_WS", 1024 * 4, NULL, 2, &xHandleWS);
```

tao-a-man, 2024-02-16:
hii @Lzw655, Can you help me check it 😍

gabsuren, 2024-07-09:
@tao-a-man  Thank you for raising the issue.

The errors `tcp_write error, errno = No more processes` and `transport_ws: Error write header` seem to be caused by resource exhaustion due to too many simultaneous tasks. To address this, ensure proper cleanup and resource management in your tasks. Limiting the number of concurrent tasks or increasing system resources might also help. Additionally, you can refer to the example below for guidance:

[ESP WebSocket Client Example](https://github.com/espressif/esp-protocols/tree/master/components/esp_websocket_client/examples/target)

Hope this helps!

closed 2024-09-09 by euripedesrocha