espressif/esp-idf #13187: ESP32 unexpected hangs with BLE+ automatic light-sleep

AhmedSalem636, 2024-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi,

I am working on a project that uses the esp32's ble power save example.

my project uses BLE with 3seconds connection interval to allow for a longer sleep time and, hence saving more power. 
the rest of the code is basically two FreeRTOS tasks:
* one runs once every 40 seconds to make some calculations " it prints the results on the monitor every time this task is called"
* the second one is suspended until we get a message over BLE 

the code was runnig just fine for 6 days and now I am unable to see my BLE device on my android BLE scanner. I thought it is the BLE module that went faulty but when I checked with my laptop serial monitor, I no longer see the esp32 printing anything which means it stopped called the periodic task as well.

my project is powered through a battery by which I can estimate how much current the esp23 is pulling. After measuring the current, it seems like the ESP draws too little current which suggests that the ESP is in a permanent sleep mode now.

AhmedSalem636, 2024-02-16:
Now, I did reset the board but I get a guru meditation error whenever I try to connect with the BLE 

here are the two error I usually get
[image]
[image]

esp-wzh, 2024-02-21:
Hi~ @AhmedSalem636 
Has your application ever accessed float type data in the ISR, if so please try enable [`FREERTOS_FPU_IN_ISR`](https://docs.espressif.com/projects/esp-idf/en/v5.1.2/esp32/api-reference/kconfig.html?highlight=freertos_fpu_in_isr#config-freertos-fpu-in-isr) option.

AhmedSalem636, 2024-03-09:
Hi,
Sorry for my late reply. 

I just tried your recommendation, it seems to make things better, but I notice it still happens.

Right now, I am testing different things " disable the automatic light sleep, trying off-the-shelf devkits instead of my custom PCB."

can bad PCB material or bad- design practice around the antenna area lead to this issue? " Since my system constantly adversities BLE services with automatic light-sleep mode"

Also, I noticed one time after applying your recommendation I got this guru meditation error with " assert failed: spinlock_acquire spinlock.h:122"

AhmedSalem636, 2024-03-23:
Hello,
Just following up. can anyone still support for this issue?
my esp32 hangs unexpectedly " stops advertising BLE and calling my FreeRTOS tasks" after some time and it does not work until I press on reset again.

esp-wzh, 2024-03-25:
> I just tried your recommendation, it seems to make things better, but I notice it still happens.

Is `coprocessor execption` still going to happen? Can you give more information

>  my esp32 hangs unexpectedly " stops advertising BLE and calling my FreeRTOS tasks" after some time and it does not work until I press on reset again.

Is there any backtrace? It would be best if you could provide a demo that reproduces this problem.

AhmedSalem636, 2024-03-25:
Hi esp-wzh,

thanks for your comment. I actually made a few changes after which the system seems to work fine till now. I am still testing it. So, I will reach out again to this thread to confirm that the problem is resolved or to seek your support again.

The changes I made are:
1- I replaced my 3v3 supply by a stronger one " this was causing a sudden hang in the system/ brownout event." 
2- regarding the guru meditation error, I changed a few configurations in the menuconfig:
* Forced the Freeartos to run on first core only
*Increased ISR stack size " not sure if it helped."
* in my code, I made sure all variables are pre-initialized with a constant value

As I said: It seems to work fine now, I will come back to tell you the result. 
Thanks

AhmedSalem636, 2024-06-02:
Hi, 
long time not commenting here. the Code seems to be stable. So I will close this thread. thanks for your help

closed 2024-06-02 by AhmedSalem636