espressif/esp-idf #13189: ESP_ZB_ON_OFF_LIGHT: Failed to initialize Zigbee stack

kotager, 2024-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32-H2

### Operating System used.

Windows

### How did you build your project?

VS Code IDE

### If you are using Windows, please specify command line type.

CMD

### Development Kit.

ESP32-H2-MINI-1

### Power Supply used.

USB

### What is the expected behavior?

I (901) ESP_ZB_LIGHT: Zigbee stack initialized
I (901) ESP_ZB_LIGHT: Start network steering
I (2611) ESP_ZB_LIGHT: Joined network successfully (Extended PAN ID: aa:98:48:01:a0:03:f7:84, PAN ID: 0x0e8b)

### What is the actual behavior?

I (459) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
W (8649) ESP_ZB_ON_OFF_LIGHT: Failed to initialize Zigbee stack (status: ESP_FAIL)

### Steps to reproduce.

Example projects were built: HA_ON_OFF_SWITCH on a C6 -> this seems to work
And HA_ON_OFF_LIGHT was build on a H2 -> this is producing the error and hangs
No alteration on the code was done

### Debug Logs.

```plain
ESP-ROM:esp32h2-20221101
Build:Nov  1 2022
rst:0x1 (POWERON),boot:0xc (SPI_FAST_FLASH_BOOT)
SPIWP:0xee
mode:DIO, clock div:1
load:0x4083cfd0,len:0xcd4
load:0x4083efd0,len:0x2da0
load:0x408460e8,len:0x1778
entry 0x4083cfd0
I (23) boot: ESP-IDF v5.1.2 2nd stage bootloader
I (24) boot: compile time Feb 15 2024 22:46:04
I (25) boot: chip revision: v0.1
I (26) boot.esp32h2: SPI Speed      : 64MHz
I (31) boot.esp32h2: SPI Mode       : DIO
I (36) boot.esp32h2: SPI Flash Size : 2MB
I (41) boot: Enabling RNG early entropy source...
I (46) boot: Partition Table:
I (50) boot: ## Label            Usage          Type ST Offset   Length
I (57) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (64) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (72) boot:  2 factory          factory app      00 00 00010000 000a2800
I (79) boot:  3 zb_storage       Unknown data     01 81 000b3000 00004000
I (87) boot:  4 zb_fct           Unknown data     01 81 000b7000 00000400
I (95) boot: End of partition table
I (99) esp_image: segment 0: paddr=00010020 vaddr=42068020 size=10d30h ( 68912) map
I (128) esp_image: segment 1: paddr=00020d58 vaddr=40800000 size=072c0h ( 29376) load
I (139) esp_image: segment 2: paddr=00028020 vaddr=42000020 size=65bc0h (416704) map
I (267) esp_image: segment 3: paddr=0008dbe8 vaddr=408072c0 size=06c48h ( 27720) load
I (281) boot: Loaded app from partition at offset 0x10000
I (281) boot: Disabling RNG early entropy source...
I (293) cpu_start: Unicore app
I (293) cpu_start: Pro cpu up.
W (302) clk: esp_perip_clk_init() has not been implemented yet
I (308) cpu_start: Pro cpu start user code
I (309) cpu_start: cpu freq: 96000000 Hz
I (309) cpu_start: Application information:
I (311) cpu_start: Project name:     light_bulb
I (317) cpu_start: App version:      1
I (321) cpu_start: Compile time:     Feb 15 2024 22:44:16
I (327) cpu_start: ELF file SHA256:  1555dcf30c41730c...
I (333) cpu_start: ESP-IDF:          v5.1.2
I (338) cpu_start: Min chip rev:     v0.0
I (343) cpu_start: Max chip rev:     v0.99
I (348) cpu_start: Chip rev:         v0.1
I (352) heap_init: Initializing. RAM available for dynamic allocation:
I (360) heap_init: At 40814690 len 00038CF0 (227 KiB): D/IRAM
I (366) heap_init: At 4084D380 len 00002B60 (10 KiB): STACK/DIRAM
I (374) spi_flash: detected chip: generic
I (377) spi_flash: flash io: dio
W (381) spi_flash: Detected size(4096k) larger than the size in the binary image header(2048k). Using the size in the binary image header.
I (395) sleep: Configure to isolate all GPIO pins in sleep state
I (401) sleep: Enable automatic switching of GPIO sleep configuration
I (408) app_start: Starting scheduler on CPU0
I (413) main_task: Started on CPU0
I (413) main_task: Calling app_main()
W (423) rmt: channel resolution loss, real=10666666
I (423) gpio: GPIO[8]| InputEn: 0| OutputEn: 1| OpenDrain: 0| Pullup: 1| Pulldown: 0| Intr:0
I (443) phy: phy_version: 211,0, 5857fe5, Nov  1 2023, 11:31:09
I (443) phy: libbtbb version: ce629d6, Nov  1 2023, 11:31:19
I (463) main_task: Returned from app_main()
I (473) ESP_ZB_ON_OFF_LIGHT: ZDO signal: ZDO Config Ready (0x17), status: ESP_FAIL
I (473) ESP_ZB_ON_OFF_LIGHT: Zigbee stack initialized
W (8663) ESP_ZB_ON_OFF_LIGHT: Failed to initialize Zigbee stack (status: ESP_FAIL)
```

xieqinan, 2024-02-29:
@kotager ,

I just attempt to use the esp-idf v5.1.2 to reprouduce your issue, but it works fine.

Could you attempt to use the below command to flash the zigbee firmware?
```
idf.py -p [PORT] erase-flash flash monitor
```

kotager, 2024-03-08:
Hi xieqinan,
Thanks for taking a look. I keep getting the following now:
[image: image.png]

xieqinan ***@***.***> ezt írta (időpont: 2024. febr. 29., Cs,
13:28):

kotager, 2024-03-09:
Hi xieqinan,I’ve just realized, that I may have a “strange” board”. (please find a photo attached)It seems missing the 3.3 power on led, also the layout is different than on the esp32 site.The exact model reference was:ESP32-C6-DevKitC-1-N4ESP32-C6 ESP32-C6-DevKitC-1-N4 Entwicklungsplatine RUIZHI ESP32-C6-DevKitC-1-N4 Modul Unterstützt WiFi Bluetooth 512KB SRAM 320KB ROM Type-C Schnittstelle 4M Kompatibel mit Arduinoamazon.deCan this cause the issue? I was not able to find a bord with this outlook on the official site.

chshu, 2024-11-13:
@kotager Please reopen if any further questions.

closed 2024-11-13 by chshu