espressif/esp-idf #13190: CMake Build - Incorrect Escape of IDF_VER

nkaaf, 2024-02-15:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Operating System used.

macOS

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

Escaped Character Sequence:

```cmake
set(COMPILE_DEFINITIONS "_GNU_SOURCE;IDF_VER=\"v5.1.2-dirty\"")
```

### What is the actual behavior?

Not Escaped Character Sequence:

```cmake
set(COMPILE_DEFINITIONS "_GNU_SOURCE;IDF_VER="v5.1.2"")
```

### Steps to reproduce.

Currently tested under macOS (M1) in Dev Container (based on Ubuntu 22.04 and latest toolchain)

* ESP-IDF 5.1.2 Installation
* Creation of Project or Component
* Build with CMake

### Build or installation Logs.

```plain
[proc] Befehl wird ausgeführt: /usr/bin/cmake --no-warn-unused-cli -DCMAKE_BUILD_TYPE:STRING=debug -DCMAKE_EXPORT_COMPILE_COMMANDS:BOOL=TRUE -DCMAKE_C_COMPILER:FILEPATH=/home/vscode/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DCMAKE_CXX_COMPILER:FILEPATH=/home/vscode/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-g++ -S/workspaces/SD -B/workspaces/SD/cmake-build-debug-vscode -G Ninja
[cmake] Not searching for unused variables given on the command line.
[cmake] -- Building ESP-IDF components for target esp32
[cmake] -- Checking Python dependencies...
[cmake] Python requirements are satisfied.
[cmake] Constraint file: /home/vscode/.espressif/espidf.constraints.v5.1.txt
[cmake] Requirement files:
[cmake]  - /home/vscode/esp/esp-idf/tools/requirements/requirements.core.txt
[cmake] Python being checked: /home/vscode/.espressif/python_env/idf5.1_py3.10_env/bin/python
[cmake] Processing 1 dependencies:
[cmake] [1/1] idf (5.1.2)
[cmake] CMake Error at /home/vscode/esp/esp-idf/tools/cmake/component.cmake:224 (message):
[cmake]   CMake Warning (dev) at build_properties.temp.cmake:8:
[cmake] 
[cmake]     Syntax Warning in cmake code at column 47
[cmake] 
[cmake]   
[cmake] 
[cmake]     Argument not separated from preceding token by whitespace.
[cmake] 
[cmake]   Call Stack (most recent call first):
[cmake] 
[cmake]     /home/vscode/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:3 (include)
[cmake] 
[cmake]   This warning is for project developers.  Use -Wno-dev to suppress it.
[cmake] 
[cmake]   
[cmake] 
[cmake]   CMake Error at /workspaces/X/CMakeLists.txt:8 (file):
[cmake] 
[cmake]     file failed to open for reading (No such file or directory):
[cmake] 
[cmake]   
[cmake] 
[cmake]       /workspaces/SD/cmake-build-debug-vscode/version
[cmake] 
[cmake]   Call Stack (most recent call first):
[cmake] 
[cmake]     /home/vscode/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:106 (include)
[cmake]     /home/vscode/esp/esp-idf/tools/cmake/scripts/component_get_requirements.cmake:124 (__component_get_requirements)
[cmake] 
[cmake]   
[cmake] 
[cmake]   
[cmake] 
[cmake] Call Stack (most recent call first):
[cmake]   /home/vscode/esp/esp-idf/tools/cmake/build.cmake:574 (__component_get_requirements)
[cmake]   /home/vscode/esp/esp-idf/tools/cmake/project.cmake:547 (idf_build_process)
[cmake]   CMakeLists.txt:19 (project)
[cmake] 
[cmake] 
[cmake] -- Configuring incomplete, errors occurred!
```

### More Information.

To be honest I do not know why this is bug is occurring right now. I have done some projects already and everything worked. But now I have my first project inside a DevContainer and this bug appears.
I have fixed the bug locally by changing the following (line 79 of esp-idf/tools/cmake/build.cmake)

```cmake
idf_build_set_property(COMPILE_DEFINITIONS "IDF_VER=\"${idf_ver}\"" APPEND)
```

to

```cmake
idf_build_set_property(COMPILE_DEFINITIONS "IDF_VER=\\\"${idf_ver}\\\"" APPEND)
```

After the change the generated line in the `build_properties.temp.cmake` looked exactly like expected (see "expected behavior")

nkaaf, 2024-02-15:
Unfortunately, the double escaping in the build process caused the following problems:

```
[build] [14/898   0% :: 0.260] Building C object esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj
[build] FAILED: esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj 
[build] /home/vscode/.espressif/tools/xtensa-esp32-elf/esp-12.2.0_20230208/xtensa-esp32-elf/bin/xtensa-esp32-elf-gcc -DESP_PLATFORM -DIDF_VER="\\\"v5.1.2-dirty\\\"" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -I/workspaces/X/cmake-build-debug-vscode/config -I/home/vscode/esp/esp-idf/components/mbedtls/port/include -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/include -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/library -I/workspaces/X/cmake-build-debug-vscode/esp-idf/mbedtls/mbedtls/library -I/home/vscode/esp/esp-idf/components/newlib/platform_include -I/home/vscode/esp/esp-idf/components/freertos/FreeRTOS-Kernel/include -I/home/vscode/esp/esp-idf/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -I/home/vscode/esp/esp-idf/components/freertos/esp_additions/include/freertos -I/home/vscode/esp/esp-idf/components/freertos/esp_additions/include -I/home/vscode/esp/esp-idf/components/freertos/esp_additions/arch/xtensa/include -I/home/vscode/esp/esp-idf/components/esp_hw_support/include -I/home/vscode/esp/esp-idf/components/esp_hw_support/include/soc -I/home/vscode/esp/esp-idf/components/esp_hw_support/include/soc/esp32 -I/home/vscode/esp/esp-idf/components/esp_hw_support/port/esp32/. -I/home/vscode/esp/esp-idf/components/esp_hw_support/port/esp32/private_include -I/home/vscode/esp/esp-idf/components/heap/include -I/home/vscode/esp/esp-idf/components/log/include -I/home/vscode/esp/esp-idf/components/soc/include -I/home/vscode/esp/esp-idf/components/soc/esp32 -I/home/vscode/esp/esp-idf/components/soc/esp32/include -I/home/vscode/esp/esp-idf/components/hal/esp32/include -I/home/vscode/esp/esp-idf/components/hal/include -I/home/vscode/esp/esp-idf/components/hal/platform_port/include -I/home/vscode/esp/esp-idf/components/esp_rom/include -I/home/vscode/esp/esp-idf/components/esp_rom/include/esp32 -I/home/vscode/esp/esp-idf/components/esp_rom/esp32 -I/home/vscode/esp/esp-idf/components/esp_common/include -I/home/vscode/esp/esp-idf/components/esp_system/include -I/home/vscode/esp/esp-idf/components/esp_system/port/soc -I/home/vscode/esp/esp-idf/components/esp_system/port/include/private -I/home/vscode/esp/esp-idf/components/xtensa/include -I/home/vscode/esp/esp-idf/components/xtensa/esp32/include -I/home/vscode/esp/esp-idf/components/lwip/include -I/home/vscode/esp/esp-idf/components/lwip/include/apps -I/home/vscode/esp/esp-idf/components/lwip/include/apps/sntp -I/home/vscode/esp/esp-idf/components/lwip/lwip/src/include -I/home/vscode/esp/esp-idf/components/lwip/port/include -I/home/vscode/esp/esp-idf/components/lwip/port/freertos/include -I/home/vscode/esp/esp-idf/components/lwip/port/esp32xx/include -I/home/vscode/esp/esp-idf/components/lwip/port/esp32xx/include/arch -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/everest/include -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -mlongcalls -Wno-frame-address  -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -O0 -g3 -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -Wno-unused-parameter -Wno-sign-compare -Wno-enum-conversion -gdwarf-4 -ggdb -Og -fmacro-prefix-map=/workspaces/X=. -fmacro-prefix-map=/home/vscode/esp/esp-idf=/IDF -fstrict-volatile-bitfields -fno-jump-tables -fno-tree-switch-conversion -DconfigENABLE_FREERTOS_DEBUG_OCDAWARE=1 -std=gnu17 -Wno-old-style-declaration -MD -MT esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj -MF esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj.d -o esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj -c /home/vscode/esp/esp-idf/components/mbedtls/mbedtls/library/aes.c
[build] <command-line>: error: missing terminating " character [-Werror]
[build] cc1: all warnings being treated as errors
```

The corresponding line in the `build.ninja` file is:

```ninja
build esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj: C_COMPILER__mbedcrypto_debug /home/vscode/esp/esp-idf/components/mbedtls/mbedtls/library/aes.c || cmake_object_order_depends_target_mbedcrypto
  DEFINES = -DESP_PLATFORM -DIDF_VER="\\\"v5.1.2-dirty\\\"" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS
  DEP_FILE = esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/aes.c.obj.d
  FLAGS = -mlongcalls -Wno-frame-address  -Wall -Wextra -Wwrite-strings -Wformat=2 -Wno-format-nonliteral -Wvla -Wlogical-op -Wshadow -Wformat-signedness -Wformat-overflow=2 -Wformat-truncation -Werror -Wmissing-declarations -Wmissing-prototypes -O0 -g3 -ffunction-sections -fdata-sections -Wall -Werror=all -Wno-error=unused-function -Wno-error=unused-variable -Wno-error=unused-but-set-variable -Wno-error=deprecated-declarations -Wextra -W [... cut ...] ts/mbedtls/mbedtls/3rdparty/p256-m -I/home/vscode/esp/esp-idf/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m
  OBJECT_DIR = esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir
  OBJECT_FILE_DIR = esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir
  TARGET_COMPILE_PDB = esp-idf/mbedtls/mbedtls/library/CMakeFiles/mbedcrypto.dir/mbedcrypto.pdb
  TARGET_PDB = esp-idf/mbedtls/mbedtls/library/libmbedcrypto.pdb
```

I will determine if the problem is related to anything in my Toolchain.

nkaaf, 2024-02-15:
I found the real problem. It was actually something in my toolchain. Sorry 😄

closed 2024-02-15 by nkaaf

pebender, 2024-06-03:
nkaaf, could you please post what the cause and solution were? thank you.

nkaaf, 2024-06-03:
@pebender Hi, if I remember correctly, the problem had nothing to do with ESP-IDF. It is exactly as stated in the CMake output
```
[cmake]   CMake Error at /workspaces/X/CMakeLists.txt:8 (file):
[cmake] 
[cmake]     file failed to open for reading (No such file or directory):
[cmake] 
[cmake]   
[cmake] 
[cmake]       /workspaces/X/cmake-build-debug-vscode/version
```
The version file referred to my project and was not part of the ESP-IDF. I guess I misplaced the file initially.

At the time I created this issue, I was focusing on the messages above the copied message, that contained some warnings in the ESP-IDF. But to be honest, I don't remember if they disappeared after I fixed my FileNotFound bug. But I think they did.

I hope this helps you :)

zhou5852, 2025-01-30:
我这边也同样出现这个问题了 主要是Clion 的cmake 中出现的

[image]

[image]

[image]

zhou5852, 2025-01-30:
[image]