espressif/esp-idf #13193: Regarding storing backtrace and retriving during reboot

sett-dozee, 2024-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I want to be able to store the backtrace in someway --> retrieve it during a reboot --> write to a file --> upload to the cloud

I am using ESP-IDF v4.3

In trying to do this, I have faced 2 issues.

I followed the steps in the [CoreDump Documentation](https://docs.espressif.com/projects/esp-idf/en/release-v4.3/esp32/api-guides/core_dump.html) and created the partition as below
```
# Name,   Type, SubType, Offset,   Size, Flags
# Note: if you change the phy_init or app partition offset, make sure to change the offset in Kconfig.projbuild
nvs,      data, nvs,     ,        0x4000,
otadata,  data, ota,     ,        0x2000,
phy_init, data, phy,     ,        0x1000,
coredump, data, coredump,,        2M
factory,  app,  factory, ,        4M,
ota_0,    app,  ota_0,   ,        3M,
ota_1,    app,  ota_1,   ,        3M,
```

When I build the code after this I get the following error
```
/home/dozee/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: dozee_next.elf section `.iram0.text' will not fit in region `iram0_0_seg'
/home/dozee/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: IRAM0 segment data does not fit.
/home/dozee/.espressif/tools/xtensa-esp32-elf/esp-2020r3-8.4.0/xtensa-esp32-elf/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld: region `iram0_0_seg' overflowed by 1056 bytes
```
I changed the setting in `idf.py menuconfig` `Compiler options --> Optimization level` from `Debug (-0g)` to `Optimize for size (-0s)` after which I am able to build the project. However when the firmware is flashed, the device seems to crash with the following message printed
```
Guru Meditation Error: Core  0 panic'ed (Unhandled debug exception). 
Debug exception reason: BREAK instr 
Core  0 register dump:
PC      : 0x400803c0  PS      : 0x00060336  A0      : 0x00000000  A1      : 0x3ffb44b0  
A2      : 0x3ffca734  A3      : 0x3ffc6a88  A4      : 0x3ffc855c  A5      : 0x00000000  
A6      : 0x00000000  A7      : 0x00000000  A8      : 0x800e7e64  A9      : 0x3ffb4050  
A10     : 0x3ffb4270  A11     : 0x3ffb4468  A12     : 0x3ffb4070  A13     : 0x3ffca98c  
A14     : 0x3ffb4070  A15     : 0x3ffb439c  SAR     : 0x00000004  EXCCAUSE: 0x00000001  
EXCVADDR: 0xffffffe0  LBEG    : 0x400013f9  LEND    : 0x4000140d  LCOUNT  : 0xfffffffa  

Backtrace:0x400803bd:0x3ffb44b0

ELF file SHA256: 0505095f6c3b48dd

Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x1b (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:4688
load:0x40078000,len:14268
load:0x40080400,len:3592
entry 0x4008065c
```
My first issue is that I am not able to debug anything from the backtrace using the command `xtensa-esp32-elf-addr2line -pfiaC -e build/dozee_next.elf Backtrace:0x400803bd:0x3ffb44b0`. However if I enable `Print coredump to UART`, and decode the ELF content I get the following output
```
espcoredump.py v0.4-dev
===============================================================
==================== ESP32 CORE DUMP START ====================

Crashed task handle: 0x3ffb4590, name: 'usbserial', GDB name: 'process 1073431952'

================== CURRENT THREAD REGISTERS ===================
[... 646 lines cut ...]
.flash.text 0x400d0020 0x13213 R XA
.iram0.data 0x4008a800 0x0 RW  
.iram0.bss 0x4008a800 0x0 RW  
.dram0.heap_start 0x3ffc0e18 0x0 RW  
.coredump.tasks.data 0x3ffbc8f4 0x168 RW 
.coredump.tasks.data 0x3ffbc4d0 0x41c RW 
.coredump.tasks.data 0x3ffbce64 0x168 RW 
.coredump.tasks.data 0x3ffbccd0 0x18c RW 
.coredump.tasks.data 0x3ffbd3d4 0x168 RW 
.coredump.tasks.data 0x3ffbd240 0x18c RW 
.coredump.tasks.data 0x3ffb7460 0x168 RW 
.coredump.tasks.data 0x3ffb72d0 0x188 RW 
.coredump.tasks.data 0x3ffb7ce4 0x168 RW 
.coredump.tasks.data 0x3ffbb0d0 0x1a8 RW 
.coredump.tasks.data 0x3ffbd540 0x168 RW 
.coredump.tasks.data 0x3ffc1b20 0x150 RW 
.coredump.tasks.data 0x3ffb7ad8 0x168 RW 
.coredump.tasks.data 0x3ffb7930 0x1a0 RW 

===================== ESP32 CORE DUMP END =====================
===============================================================
Done!
```

Can someone please help me with the above issues or tell another method to capture the backtrace?

igrr, 2024-02-16:
> but the backtrace from the current thread stack in the decoded ELF text does not match the backtrace in the logs

This issue has been fixed in IDF v4.4.x and later (https://github.com/espressif/esp-idf/issues/6124), please try updating to a new version.

> This is the first problem, the code is crashing when compiled with Compiler option = Optimize for size and I am not able to understand what the problem could be causing #0  _DoubleExceptionVector () at /home/dozee/esp/esp-idf/components/freertos/port/xtensa/xtensa_vectors.S:455

I recommend checking the stack size of your `usbserial` task, it could be that you are overflowing the stack, which leads to heap corruption and the subsequent crash inside the scheduler.

sett-dozee, 2024-02-16:
Thanks for the quick response

sett-dozee, 2024-02-16:
The USB serial task has a stack size of 6 KB and when I build the code with `Compiler options = Debug`, this issue is not seen

closed 2025-03-08 by Alvin1Zhang