espressif/esp-idf #13194: Upgrading to 3,0,0 and having issues!

dagnall53, 2024-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

I am trying to make my code acceptable to 3.0.0 and am hitting issues I do not understand:
Specifically: I am now down to some apparently related  issues where I do not seem to be "seeing" some functions at compile . 

1.  error: 'gpio_matrix_out' was not declared in this scope; did you mean 'gpio_iomux_out'? 1891 |   gpio_matrix_out(TX_PIN_0, 0x100, false, false);
2. error: 'gpio_matrix_in' was not declared in this scope   132 |         gpio_matrix_in(RxPin,CAN_RX_IDX,0);
3. error: 'gpio_pad_select_gpio' was not declared in this scope; did you mean 'esp_rom_gpio_pad_select_gpio'?
  133 |         gpio_pad_select_gpio(RxPin);

I have searched the new IDF and gpio_matrix_out appears to be defined in gpio.h  so why does my code not find it when the previous IDF worked fine? - I have board with ESP32-WROOM-32, and am compiling for "ESP32 DEV board"

There will probably be more issues.. but this is enough to stop the compile and thoroughly confuse me!!! 
Many thanks
Dagnall

dagnall53, 2024-02-16:
@sudeep-mohanty  Hi, if I understand the question, NO, I am using the Expressive IDF V3.0.0alpha3 . Not the Arduino ESp32 boards Which only goes "up" to 2,0.13..  
What is confusing me is that the IFD seems to "not know" about these functions, but I have checked and I think they exist!
Its possible IDF V3.0.0alpha3 does not somehow call or include the source - but the previous versions did!

dagnall53, 2024-02-17:
@sudeep-mohanty Thanks.. 
Is it possible that in previous versions rom/gpio.h was "automagically" included in "something".. but that has been "corrected"? 
for the simple.. (ie ME!) do I need to #include <rom/gpio.h>  I have not had to use the path before!
Currently I have 
````
#include <ESPmDNS.h>
#include <WiFi.h>
#include <WiFiUdp.h>
#include <EEPROM.h>
#include <esp_WiFi.h>
#include <WebServer.h>
#include <Update.h>
#include <esp_now.h>
````

sudeep-mohanty, 2024-02-17:
@dagnall53 While I haven't evaluated how `rom/gpio.h` was indirectly included from one of these header files prior to IDF 3.0, it is quite possible that you encounter such breaking changes when you upgrade to a major release. We actively try to reduce such dependencies between IDF components by removing unnecessary header file inclusions. Usually, we document such breaking changes in the release notes but there are times when some of them slip through our internal reviews and show up in user applications. Such is the case here and you would have to include `rom/gpio.h` in your application to compile with the functions you need.

dagnall53, 2024-02-17:
@sudeep-mohanty 
Thanks. I get that these things can get broken / missed!
.. But I could you confirm that I was correct in the formulation of the **#include   <rom/gpio.h>**
or should it be **#include < gpio >**   or ... ? 
I suspect you were overestimating my competence! :-)

sudeep-mohanty, 2024-02-17:
That's correct. You need to have '#include "rom/gpio.h"'in your application source file.

dagnall53, 2024-02-18:
@sudeep-mohanty Ahh.. some more digging and I find (2.0.14) (C:\Users\admin\AppData\Local\Arduino15\packages\esp32\hardware\esp32\2.0.14\tools\sdk\esp32\include\esp32\include\rom\gpio.h)
```
#warning rom/gpio.h is deprecated, please use esp32/rom/gpio.h instead
#include "esp32/rom/gpio.h"
```
Perhaps that has been forgotten in V3 when updating and some elements reference the wrong (old reference)  now?

sudeep-mohanty, 2024-02-19:
Hello @dagnall53, Looking at your latest comment, it strikes me that you are probably upgrading your Arduino core from 2.0.14 to 3.0.0. This, in-turn, would mean that you are moving your IDF versions from 4.4 to 5.1. v5.1 of IDF supports both `#include "rom/gpio.h" or `#include "esp32/rom/gpio.h"` (essentially the same file) to have access to these GPIO matrix APIs. You could also use the API functions in [esp_rom_gpio.h](https://github.com/espressif/esp-idf/blob/cbce221e88d52665523093b2b6dd0ebe3f1243f1/components/esp_rom/include/esp_rom_gpio.h#L83) which provide target-agnostic APIs for the APIs you are working with. Apologies for earlier as I botched up version changes in my head!

dagnall53, 2024-02-19:
Thanks. I will have to investigate that more.. !
Its all a bit opaque to me right now.
It started with "just" a "simple"  gpio_matrix_out(TX_PIN_0, 0x100, false,
false);
that had been used to overcome another issue that started this all off!
And then I discovered that the NMEA2000 library also seems to suffer!
O well, I will proceed one hack at a time.

I really appreciate your help and guidance!
Dagnall

sudeep-mohanty, 2024-03-20:
Since there is no bug to be fixed, closing this issue. Please feel free to reopen a new one in case of further support. Thanks!

closed 2024-03-20 by sudeep-mohanty