espressif/esp-idf #13195: BLE wifi provisioning with BLE periodic sync issue

Sotung96, 2024-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### General issue report

Hi all, I'm trying to use [wifi provisioning manager example](https://github.com/espressif/esp-idf/tree/master/examples/provisioning/wifi_prov_mgr) for testing BLE (using BLE to provisioning). Also, I want to add [Ble Periodic sync example](https://github.com/espressif/esp-idf/tree/master/examples/bluetooth/nimble/ble_periodic_sync) into wifi provisioning example. However, I cannot do any provisioning after I enable extended advistering. How can I make 2 example together for test BLE? Thanks

rahult-github, 2024-02-28:
Hi @espressif-bot  , 

 For integrating the ble periodic advertising, i think you need to refer  the BLE Periodic adv example, instead of Ble periodic sync example, since you will need to advertise.   
 
 As for provisioning not happening, this may be because if you are using Esp Wifi prov application, then currently it doesn't support periodic adv report parsing. We can take it up as a feature request and plan add it going ahead though.

Sotung96, 2024-03-19:
@rahult-github Yes please because I am trying to put ble provisioning and periodic adv into my project. Thanks for your help.