espressif/esp-idf #13196: IDF 5.2 fails to install Python venv

readmodifywrite, 2024-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2 11eaf41b37267ad7709c0899c284e3683d2f0b5e

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

IDF 5.2 should.... work? Lol.

### What is the actual behavior?

ERROR: /home/jeremy/.espressif/python_env/idf5.2_py3.8_env/bin/python doesn't exist! Please run the install script or "idf_tools.py install-python-env" in order to create it

This is after erasing ~/.espressif and running install.sh.

v5.2 doesn't create idf5.2_py3.8_env, so nothing works.  Can't even run a clean command.

It does however, install idf5.1_py3.8_env.

It looks like renaming it to 5.2 instead of 5.1 gets it to work.

I'm using the actual v5.2 tag.

Did this work in your CI system?  Why would it install differently on my machine vs the CI environment?

### Steps to reproduce.

Install IDF 5.2.  That's it.

igrr, 2024-02-16:
> Did this work in your CI system?

It did successfully install when building the Docker image: https://github.com/espressif/esp-idf/actions/runs/7911595216/job/21595996329, and it works in our internal CI as well.

Can you please try running `git describe` inside your copy of IDF? Does it say `v5.2` or something else?

readmodifywrite, 2024-02-16:
v5.2

I checked out the actual tag.

It looks like a lot of commands are still looking for the 5.1 venv, and some are looking for 5.2.

So the fix is to copy the 5.1 the installer creates into 5.2, and keep the old 5.1.

fhrbata, 2024-02-16:
Hello @readmodifywrite ,

just a thought. Isn't it possible that you are running the `install.sh` within `5.1` env. activated.  Meaning the `IDF_PYTHON_ENV_PATH` is actually set to `5.1` and it's used by the `idf_tool.py`?

I can reproduce this with

1. checkout v5.1 tag and run `install.sh`
2. activate 5.1 with `export.sh`
3. checkout v5.2
4. delete `~/.espressif`
5. run install.sh

In 2. the `IDF_PYTHON_ENV_PATH` is set to 5.1 and it's used in 5.  Just a quick thought that might help.

closed 2024-03-06 by espressif-bot