espressif/esp-idf #13197: v5.2 bootloader crashes on startup

readmodifywrite, 2024-02-16:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.2 11eaf41b37267ad7709c0899c284e3683d2f0b5e

### Espressif SoC revision.

ESP32-C3

### Operating System used.

Linux

### How did you build your project?

Command line with CMake

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

The chip should boot up to the user application.

### What is the actual behavior?

The bootloader crashes on startup.

### Steps to reproduce.

Run IDF tag v5.2 on the C3.

### Debug Logs.

```plain
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0xc (RTC_SW_CPU_RST),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x403808ec
SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x23e0
load:0x403cc710,len:0xe4c
load:0x403ce710,len:0x354c
entry 0x403cc71a
I (24) boot: ESP-IDF v5.2 2nd stage bootloader
I (24) boot: compile time Feb 16 2024 10:12:36
D (24) bootloader_flash: non-XMC chip detected by SFDP Read (00), skip.
D (29) bootloader_flash: mmu set block paddr=0x00000000 (was 0xffffffff)
I (36) boot: chip revision: v0.4
D (40) qio_mode: Probing for QIO mode enable...
D (45) qio_mode: Raw SPI flash chip id 0x5e4016
D (49) qio_mode: Manufacturer ID 0x5e chip ID 0x4016
I (54) qio_mode: Enabling default flash chip QIO
D (60) qio_mode: Initial flash chip status 0x2
D (64) qio_mode: QIO mode already enabled in flash
[... 135 lines cut ...]
3fcde4c0: 0x00000000 0x00000000 0x00010000 0x00180000 0x00190000 0x00180000 0x00000000 0x00000000
3fcde4e0: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde500: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde520: 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde540: 0x00000000 0x00000000 0x00000002 0x00000000 0x600c5200 0x600c5200 0xdbb80087 0x4004a296
3fcde560: 0x00000000 0x00000000 0x3fcde668 0x00000000 0xffffffff 0x00000000 0x00010000 0x2f0203e9
3fcde580: 0x403cc71a 0x403ce710 0x0000354c 0x000000ee 0x03030005 0x0000c700 0x01000000 0x3c0067b0
3fcde5a0: 0x7533885e 0xe608c91c 0x1d5ebe25 0xfcb99d89 0x50412066 0x081ea972 0xbd990896 0xf99cb28f
3fcde5c0: 0xf9e235c5 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
3fcde5e0: 0x00000000 0x3fcde700 0x00000101 0x00000002 0xe608c91c 0x1d5ebe25 0xfcb99d89 0x50412066
3fcde600: 0x081ea972 0xbd990896 0xf99cb28f 0xf9e235c5 0x00000000 0x00000000 0x00000000 0x00000000
3fcde620: 0x00000000 0x00000000 0x00000000 0x00000000 0x600087b7 0x669d5bb8 0x9b5d0532 0x5bb8dbb8
3fcde640: 0x76e58d75 0x8f7516fd 0xdba88d59 0x67135bb8 0xdbb80087 0x00008082 0x00000000 0x36000000
3fcde660: 0x00000080 0x00000000 0x00000000 0x803d0300 0x00000000 0x00000000 0x00000000 0x00000000

ELF file SHA256: 

Rebooting...
```

### More Information.

This is fatal for v5.2.  This commit should not be tagged with a broken bootloader.  Please delete the tag so no one else runs in to this.

What commit (I want the exact hash) on 5.2 actually works?

readmodifywrite, 2024-02-16:
This issue persists on release/v5.2 as well: cc649ea869b8c24d19ba2d94c4d3f1146cb2947b

*DO NOT USE THIS RELEASE*

Absolute fail y'all.  I'm going back to 5.1 - please do not suggest updating to 5.2 to deal with the bugs in 5.1 until 5.2 is able to *boot up the chip*.

readmodifywrite, 2024-02-16:
I can confirm that reverting IDF to v5.1 is a viable fix.

igrr, 2024-02-16:
Sorry for the bad experience with the version upgrade @readmodifywrite! There haven't been many changes since v5.2-beta1 release, and so far we didn't get feedback that it's absolutely broken, so we went ahead with the final release. In our internal CI, all IDF examples we test on real hardware also worked fine. However sometimes there are cases which are not covered by our CI, usually related to a particular combination of sdkconfig settings. Hopefully we can figure out what is causing the issue in your case quickly.

According to the log, the crash happens during app startup (cpu_start line). Could you please clarify if you are
- using IDF v5.2 bootloader with an IDF v5.1 app binary?
- or building both the app and the bootloader with v5.2? If yes, please also attach you application's sdkconfig file.

Additionally, could you please use `idf.py monitor` to get the decoded backtrace? This will help us identify the issue.

readmodifywrite, 2024-02-16:
Everything is built on 5.2.  It never gets to the app binary.  I'm loading over serial with esptool.

I don't have the sdkconfig at the moment, I've already reverted to 5.1 and am now trying to unfuck the mDNS build which refuses to work without the useless component manager.

I'm sorry I can't be more helpful, but it's just whack-a-mole with broken behavior in the IDF.  I've got a ton of issues with the MQTT client I'm trying to sort out too (was hoping 5.2 would clean some of this up, apparently it won't).

There is just too much instability, unnecessary complexity, and change for the sake of change for me to keep up.  It's a full time job just to workaround bugs in the IDF, and I'm just a solo dev.  At some point I need to write my actual app code so I have a product to sell!

igrr, 2024-02-16:
> Everything is built on 5.2. It never gets to the app binary.

The `I (825) cpu_start: Unicore app` line indicates that it does get to the app binary, although it crashes soon after, before it gets to your app_main function. (If it crashed in the bootloader, the panic output would also look differently.)

If you manage to share the sdkconfig file and provide the decoded backtrace in the future, it would be very helpful for diagnosing this issue.

readmodifywrite, 2024-02-16:
FWIW, y'all really do an amazing job of support here.  No other chip vendor even bothers to try.

I'll try and dig a bit more on 5.2 next week - I need to circle back to my MQTT stuff for now.  I did get the mDNS sorted so that's something.

andylinpersonal, 2024-02-17:
Seems like it failed when the default console is configured as USB Serial JTAG on ESP32-C3.

sudeep-mohanty, 2024-02-19:
Hello @readmodifywrite, 
I tried a few default IDF examples with v5.2 but I do not encounter this issue. Even when USB-serial-JTAG is enabled, the bootup looks fine and I am able to enter the user app as intended. Could you please provide some more details about your application and maybe the `sdkconfig` file? It might help us in narrowing down to the configuration which is causing this problem. Thanks!

andylinpersonal, 2024-02-20:
Here is a MWE for this issue

IDF version: v5.2-166-ge0944287f3 (e0944287f33b348f9ccb7026bd32517d0f6b3e56)
Target:      esp32-c3 v0.4

main/main.c
```c
#include <freertos/FreeRTOS.h>
#include <stdio.h>

void app_main(void) {
  while (1) {
    fputs("hello world!\n", stderr);
    vTaskDelay(1000);
  }
}
```

main/CMakeLists.txt
```cmake
idf_component_register(SRCS "main.c")
```

CMakeLists.txt
```cmake
cmake_minimum_required(VERSION 3.16)
include($ENV{IDF_PATH}/tools/cmake/project.cmake)

project(tester)
```

sdkconfig.defaults
```
# C3 stays in the reset loop with this option on
CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y
```

Log messages
```
 ESP-ROM:esp32c3-api1-20210207
 Build:Feb  7 2021
 rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
 Saved PC:0x4038061e
 SPIWP:0xee
 mode:DIO, clock div:1
 load:0x3fcd5820,len:0x1738
 load:0x403cc710,len:0xaf0
 load:0x403ce710,len:0x2e3c
 entry 0x403cc71a
 I (15) boot: ESP-IDF v5.2-166-ge0944287f3 2nd stage bootloader
 I (15) boot: compile time Feb 20 2024 19:21:21
 I (15) boot: chip revision: v0.4
 I (16) boot.esp32c3: SPI Speed      : 80MHz
 I (16) boot.esp32c3: SPI Mode       : DIO
 I (16) boot.esp32c3: SPI Flash Size : 2M I (16) boot: Enabling RNG early entropy source...
 I (16) boot: Partition Table:
 I (16) boot: ## Label            Usage          Type ST Offset   Length
 I (17) boot:  0 nvs              WiFi data        01 02 00009000 00006000
 I (17) boot:  1 phy_init         RF data          01 01 0000f000 00001000
 I (17) boot:  2 factory          factory app      00 00 00010000 00100000
 I (17) boot: End of partition table
 I (18) esp_image: segment 0: paddr=00010020 vaddr=3c020020 size=09380h ( 37760) map
 I (24) esp_image: segment 1: paddr=000193a8 vaddr=3fc8a800 size=011cch (  4556) load
 I (25) esp_image: segment 2: paddr=0001a57c vaddr=40380000 size=05a9ch ( 23196) load
 I (30) esp_image: segment 3: paddr=00020020 vaddr=42000020 size=14770h ( 83824) map
 I (44) esp_image: segment 4: paddr=00034798 vaddr=40385a9c size=04cbch ( 19644) load
 I (51) boot: Loaded app from partition at offset 0x10000
 I (51) boot: Disabling RNG early entropy source...
 I (51) cpu_start: Unicore app
 Guru Meditation Error: Core  0 panic'ed (Load access fault). Exception was unhandled.
[... 36 lines cut ...]
 3fcde540: 0x00000000 0x00000000 0x00000000 0x00000000 0x600c5200 0x600c5200 0x00000080 0x4004a296
 3fcde560: 0x00000000 0x00000000 0x3fcde668 0x00000000 0xffffffff 0x00000000 0x00010000 0x1f0203e9
 3fcde580: 0x403cc71a 0x403ce710 0x00002e3c 0x000000ee 0x03030005 0x0000c700 0x01000000 0x3c0050a0
 3fcde5a0: 0x7533885e 0xcf496ecd 0x93252fce 0x6fead1bd 0x6b88d317 0xfc76a269 0x3255d380 0xa67982e5
 3fcde5c0: 0x84eea0c1 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000 0x00000000
 3fcde5e0: 0x00000000 0x3fcde700 0x00000101 0x00000002 0xcf496ecd 0x93252fce 0x6fead1bd 0x6b88d317
 3fcde600: 0xfc76a269 0x3255d380 0xa67982e5 0x84eea0c1 0x00000000 0x00000000 0x00000000 0x00000000
 3fcde620: 0x00000000 0x00000000 0x00000000 0x00000000 0x8f7516fd 0xdba88d59 0x67135bb8 0xdbb80087
 3fcde640: 0x00008082 0x00000000 0x00000000 0xe9000000 0x00000080 0x00000000 0x00000000 0x00000000
 3fcde660: 0x00000000 0x00000000 0x00000000 0x00850200 0x00000000 0x00000000 0x00000000 0x00000000

 ELF file SHA256:

 Rebooting...
 ESP-ROM:esp32c3-api1-20210207
 Build:Feb  7 2021
 rst:0xc (RTC_SW_CPU_RST),boot:0xc (SPI_FAST_FLASH_BOOT)
 Saved PC:0x4038061e
 SPIWP:0xee
 mode:DIO, clock div:1
 load:0x3fcd5820,len:0x1738
 load:0x403cc710,len:0xaf0
 load:0x403ce710,len:0x2e3c
 entry 0x403cc71a
```

okhsunrog, 2024-02-22:
Updated to 5.2 and killed 4 hours trying to figure out why my app won't boot. That we not very fun. I confirm the issue, turn out it happens to me with USB-Serial as default output. esp32-c3 v0.3, linux

ESP-Marius, 2024-02-23:
We've also confirmed that there is indeed an issue with using `CONFIG_ESP_CONSOLE_USB_SERIAL_JTAG=y` on v5.2 and are working on fixing it.

Thanks for the feedback here which helped us find the issue and sorry for all inconvenience this has caused you guys when trying to upgrade to v5.2!

okhsunrog, 2024-02-27:
Any updates on this issue?

FL0WL0W, 2024-02-27:
> Any updates on this issue?

Seems to be fixed on latest dev branch

ESP-Marius, 2024-02-28:
Issue was fixed on master with 6768805d20cbd026b7fccf9d7adb49991d6b66a2, but unfortunately that introduced a regression with usb_otg console. My colleague is working on fixing that, and as soon as it is working properly we will merge both fixes to v5.2.

RavenSystem, 2024-02-28:
FYI: esp32c3 issue is present too with `CONFIG_ESP_CONSOLE_NONE=y`.

RavenSystem, 2024-03-07:
Bug is still present in v5.2.1 when `CONFIG_ESP_CONSOLE_NONE=y` is used.

RavenSystem, 2024-03-07:
When `CONFIG_ESP_CONSOLE_NONE=y`, the option `CONFIG_ESP_ROM_USB_SERIAL_DEVICE_NUM=3` is added automatically. This should be set to `-1`.

AxelLin, 2024-03-12:
> FYI: esp32c3 issue is present too with `CONFIG_ESP_CONSOLE_NONE=y`.

@ESP-Marius @mythbuster5 
Is this fixed for esp32c3?

ESP-Marius, 2024-03-12:
Yes, this should be fixed as per the referenced commit. Seems like the issue didnt close automatically.

If anyone is still seeing these issues then please feel free to re-open this ticket.

closed 2024-03-12 by ESP-Marius

readmodifywrite, 2024-04-23:
Thanks for the quick work on this one!

sailinglove, 2024-07-15:
@ESP-Marius @mythbuster5

Sorry for bothering you, but I'm still encountering a similar issue after upgrading my IDF to [`v5.3-rc1`](https://github.com/espressif/esp-idf/releases/tag/v5.3-rc1).

I'm using the `example_connect()` function from the protocol_examples_common library to connect to WiFi by providing the SSID and password through the idf.py monitor's serial console. However, in IDF [`v5.2-rc1`](https://github.com/espressif/esp-idf/releases/tag/v5.2-rc1), changing the _Channel for console output_ to `USB Serial/JTAG controller` resulted in a panic abort. I found this issue and upgraded my IDF to [`v5.3-rc1`](https://github.com/espressif/esp-idf/releases/tag/v5.3-rc1), which should have resolved the issue with the commit https://github.com/espressif/esp-idf/commit/6768805d20cbd026b7fccf9d7adb49991d6b66a2.

Despite this, my program still experiences a panic abort, although it now progresses to a later stage. I noticed that the commit message for https://github.com/espressif/esp-idf/commit/6768805d20cbd026b7fccf9d7adb49991d6b66a2 mentions "Fix wrong serial number that has been parsed to rom...", but the debug log before the panic abort shows an error indicating "uart_num error."

I'm quite confused and would appreciate your help in resolving this issue. I've attached the debug logs below for your reference.

Thanks a lot!

Debug Logs:

```
ESP-ROM:esp32c3-api1-20210207
Build:Feb  7 2021
rst:0x15 (USB_UART_CHIP_RESET),boot:0xd (SPI_FAST_FLASH_BOOT)
Saved PC:0x400495da
0x400495da: ets_sha_process in ROM

SPIWP:0xee
mode:DIO, clock div:1
load:0x3fcd5820,len:0x1738
load:0x403cc710,len:0xb08
load:0x403ce710,len:0x2e90
entry 0x403cc71a
I (24) boot: ESP-IDF v5.3-rc1 2nd stage bootloader
I (24) boot: compile time Jul 15 2024 08:10:11
I (25) boot: chip revision: v0.4
I (25) boot.esp32c3: SPI Speed      : 80MHz
I (25) boot.esp32c3: SPI Mode       : DIO
I (25) boot.esp32c3: SPI Flash Size : 4MB
I (25) boot: Enabling RNG early entropy source...
I (25) boot: Partition Table:
I (25) boot: ## Label            Usage          Type ST Offset   Length
I (26) boot:  0 nvs              WiFi data        01 02 00009000 00006000
I (26) boot:  1 phy_init         RF data          01 01 0000f000 00001000
I (26) boot:  2 fctry            WiFi data        01 02 00010000 00006000
I (27) boot:  3 factory          factory app      00 00 00020000 00352000
I (27) boot:  4 storage          Unknown data     01 82 00372000 00064000
I (27) boot: End of partition table
I (27) esp_image: segment 0: paddr=00020020 vaddr=3c110020 size=79d84h (499076) map
I (106) esp_image: segment 1: paddr=00099dac vaddr=3fc97200 size=03668h ( 13928) load
I (109) esp_image: segment 2: paddr=0009d41c vaddr=40380000 size=02bfch ( 11260) load
I (111) esp_image: segment 3: paddr=000a0020 vaddr=42000020 size=107e84h (1080964) map
I (280) esp_image: segment 4: paddr=001a7eac vaddr=40382bfc size=144f0h ( 83184) load
I (296) esp_image: segment 5: paddr=001bc3a4 vaddr=50000200 size=00004h (     4) load
I (304) boot: Loaded app from partition at offset 0x20000
I (304) boot: Disabling RNG early entropy source...
I (305) cpu_start: Unicore app
I (313) cpu_start: Pro cpu start user code
I (313) cpu_start: cpu freq: 160000000 Hz
I (313) app_init: Application information:
I (314) app_init: Project name:     alarm-button
I (314) app_init: App version:      782191b-dirty
I (314) app_init: Compile time:     Jul 15 2024 08:09:59
I (314) app_init: ELF file SHA256:  51f47db5a...
I (314) app_init: ESP-IDF:          v5.3-rc1
I (314) efuse_init: Min chip rev:     v0.3
I (314) efuse_init: Max chip rev:     v1.99
I (314) efuse_init: Chip rev:         v0.4
I (315) heap_init: Initializing. RAM available for dynamic allocation:
I (315) heap_init: At 3FCA7770 len 00018890 (98 KiB): RAM
I (315) heap_init: At 3FCC0000 len 0001C710 (113 KiB): Retention RAM
I (315) heap_init: At 3FCDC710 len 00002950 (10 KiB): Retention RAM
I (316) heap_init: At 50000204 len 00001DE4 (7 KiB): RTCRAM
I (317) spi_flash: detected chip: generic
I (317) spi_flash: flash io: dio
I (317) sleep: Configure to isolate all GPIO pins in sleep state
[... 73 lines cut ...]
MSTATUS : 0x00001881  MTVEC   : 0x40380001  MCAUSE  : 0x00000007  MTVAL   : 0x00000000
0x40380001: _vector_table at /home/ivanx/esp/esp-idf-v5.3-rc1/components/riscv/vectors_intc.S:54

MHARTID : 0x00000000

Backtrace:

/bin/zsh: /home/ivanx/miniconda3/lib/libtinfo.so.6: no version information available (required by /bin/zsh)
panic_abort (details=details@entry=0x3fcaa2ec "abort() was called at PC 0x403899db on core 0") at /home/ivanx/esp/esp-idf-v5.3-rc1/components/esp_system/panic.c:463
463         *((volatile int *) 0) = 0; // NOLINT(clang-analyzer-core.NullDereference) should be an invalid operation on targets
#0  panic_abort (details=details@entry=0x3fcaa2ec "abort() was called at PC 0x403899db on core 0") at /home/ivanx/esp/esp-idf-v5.3-rc1/components/esp_system/panic.c:463
#1  0x403899e6 in esp_system_abort (details=details@entry=0x3fcaa2ec "abort() was called at PC 0x403899db on core 0") at /home/ivanx/esp/esp-idf-v5.3-rc1/components/esp_system/port/esp_system_chip.c:92
#2  0x403929c6 in abort () at /home/ivanx/esp/esp-idf-v5.3-rc1/components/newlib/abort.c:38
#3  0x403899de in _esp_error_check_failed (rc=<optimized out>, file=file@entry=0x3c11a484 "", line=line@entry=25, function=function@entry=0x3c177c94 <__func__.0> "", expression=expression@entry=0x3c11a430 "") at /home/ivanx/esp/esp-idf-v5.3-rc1/components/esp_system/esp_err.c:49
[... cut ...]
#7  0x420090b4 in ws_prerequisities () at ../main/app_ws.c:122
#8  0x42007a22 in app_main () at ../main/main.c:56
#9  0x42106cc8 in main_task (args=<error reading variable: value has been optimized out>) at /home/ivanx/esp/esp-idf-v5.3-rc1/components/freertos/app_startup.c:208
#10 0x4038ac30 in vPortTaskWrapper (pxCode=<optimized out>, pvParameters=<optimized out>) at /home/ivanx/esp/esp-idf-v5.3-rc1/components/freertos/FreeRTOS-Kernel/portable/riscv/port.c:255
ELF file SHA256: 51f47db5a

Rebooting...
```

ESP-Marius, 2024-07-16:
@sailinglove, I think this is unrelated to the bug reported earlier, as far as I can see `CONFIG_EXAMPLE_LOCAL_SERVER_URL_FROM_STDIN` has never supported other console inputs than UART.

It will try to initialize an UART driver using `CONFIG_ESP_CONSOLE_UART_NUM` as the port number, which will fail due to `-1` being an invalid port.

sailinglove, 2024-07-16:
@ESP-Marius First of all thanks for the quick response! I looked again into the sdkconfig and found that `CONFIG_ESP_CONSOLE_UART_NUM` was indeed set to `-1`, though I didn't see `CONFIG_EXAMPLE_LOCAL_SERVER_URL_FROM_STDIN` anywhere, I think you meant `CONFIG_EXAMPLE_WIFI_SSID_PWD_FROM_STDIN`?

Anyway, I am just trying to provide wifi ssid and password through `idf.py monitor` console, but using UART only got me `Writing to serial is timing out. Please make sure that your application supports an interactive console and that you have picked the correct console for serial communication.` Am I missing some settings?