espressif/esp-idf #13198: UDP sendto return ENOMEM when a station disconnected.

anmnmnly, 2024-02-17:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.1.2

### Espressif SoC revision.

ESP32

### Operating System used.

Windows

### How did you build your project?

Eclipse IDE

### If you are using Windows, please specify command line type.

None

### Development Kit.

Custom board

### Power Supply used.

USB

### What is the expected behavior?

There are three devices A, B and C, where A is the AP, B and C are the stations. Device A sends data to the B and C using UDP. During the transmiting, if B disconnected or powered off, the sendto(device C) return OK.

### What is the actual behavior?

The sendto(device C) return ENOMEM for a long time (approximate 3 seconds).

### Steps to reproduce.

There are three devices A, B and C, where A is the AP, B and C are the stations. Device A sends data to the B and C using UDP. During the transmiting, disconnect or powered off the device B.

### Debug Logs.

```plain
none.
```

### More Information.

none.

MaxwellAlan, 2024-02-19:
Hi @anmnmnly 

Briefly explain the reason:
Currently, the caching management in SoftAP is relatively simple due to memory limitations. Therefore, the total number of shared caches for multiple different connections is limited. When a certain STA(B) connection is disconnected, if there are packets from that STA(B) cached on the AP side, SoftAP continues to cache packets for that STA(B) because it is unaware of the departure of the STA(B) in a timely manner. Consequently, this causes the packets intended for another STA(C) to be delayed until the packets related to STA(B) time out, resulting in a situation of "no memory" occurring within a short period of time.

There are some methods to try:
1.Increase the number of WIFI TX buffers as much as possible while ensuring sufficient application memory.
2.Try disabling modem sleep for STA (B&C) using esp_wifi_set_ps(0).

anmnmnly, 2024-02-19:
Hello @MaxwellAlan ,

Thank you for your explanation and methods to solve this issue. I tried the methods you suggested, as below:
1. Modify Wi-Fi->Max number of WiFi dynamic TX buffers to 128, the range is 1 to 128.
2. On STA side, call esp_wifi_set_ps(0).
But it doesn't work, the issue still exist. I don't confirm if the SoftAP needs to send massive data to STAs (192000 bytes/second). 

I tried to UDP multicast that can solve this issue, but the multicast can only send 96000 bytes/second. I don't know if there are some settings that can increase the sending capacity.

MaxwellAlan, 2024-02-20:
Hi @anmnmnly 

> On STA side, call esp_wifi_set_ps(0).
But it doesn't work, the issue still exist. I don't confirm if the SoftAP needs to send massive data to STAs (192000 bytes/second).

While the STA is not sleep, the AP still not send data to STA quickly enough, which may related with the interference, could you try this test in shield box?

> I tried to UDP multicast that can solve this issue, but the multicast can only send 96000 bytes/second. I don't know if there are some settings that can increase the sending capacity.

Yeah, UDP multicast which means these ptks in WiFi layer will not retransmit ( unicast pkts will retrans 32 times :) ), but also it only can using 1Mbps(125KB/s) data rate to send, so can not meet 192KB/s.

You can try with below method:
Disable 11b rates, which means the lowest data rates will be 6Mbps : https://github.com/espressif/esp-idf/blob/f68c131e5603feca21659e92ad85f0c3369692fe/components/esp_wifi/include/esp_wifi.h#L1318C1-L1319C1

anmnmnly, 2024-03-01:
@MaxwellAlan 

Thank for your kindly reply.

Disable 11b rates is effective, but it cannot completely solve my issue. It seems that decreasing the number of data is the most suitable method in my project. 

thanks agin.

Sherry616, 2024-04-02:
Thanks for reporting, we will close this issue. Feel free to reopen if have more updates.

closed 2024-04-02 by Sherry616

anmnmnly, 2024-04-26:
@MaxwellAlan 

Hi MaxwellAlan,
"Yeah, UDP multicast which means these ptks in WiFi layer will not retransmit ( **unicast pkts will retrans 32 times** :) ), but also it only can using 1Mbps(125KB/s) data rate to send, so can not meet 192KB/s."

"**unicast pkts will retrans 32 times**", Can I modify the retrans count to another?

MaxwellAlan, 2024-04-26:
> @MaxwellAlan
>
> Hi MaxwellAlan, "Yeah, UDP multicast which means these ptks in WiFi layer will not retransmit ( **unicast pkts will retrans 32 times** :) ), but also it only can using 1Mbps(125KB/s) [...]

Not supported.