espressif/esp-idf #13199: Feature request: WiFi provisioning component should allow to save credentials for more than 1 access point

idea--list, 2024-02-17:
### Is your feature request related to a problem?

Tried the wifi_provisioning component and it perfectly fits the sceneraio when we ALWAYS want to connect to the same access point. This fits perfectly for stationary devices based on an ESP32.

However for mobile devices that will be carried around by the user to several locations, it always needs the user to take some action (like pressing a button, etc.) to delete the credentials and then redo provisioning from scratch. Which is really cumbersome on a daily basis.

### Describe the solution you'd like.

I would like the wifi_provisioning component being able to save credentials for several access points. Pretty much like in WiFi>>WiFiMulti example bundled into arduino-esp32.

Took a quick view at wifi_provisioning component's code as well as on that of the ESP BLE/ESP AP Prov applications' code for android but could not identify any parts that would have catched my eyes being relevant for such a feature.

Harshal5, 2024-03-20:
Hello @idea--list,

I think this could be handled by the application rather than increasing the complexity of the provisioning component. There might be cases where credentials might not be stored in the NVS but could be retrieved for some other sources (say cloud), so I think storing and retrieving the WiFi credentials must be some user-controlled policy.

Thanks!

Harshal5, 2024-04-22:
@idea--list Did the above example suffice your use case? Was it helpful?

closed 2024-05-27 by mahavirj