espressif/esp-idf #13200: HTTP API for accessing response information

otaxhu, 2024-02-17:
### Is your feature request related to a problem?

I'm trying to implement a "middleware" (requests handlers that comes before the actual handler) architecture.

And I'm trying to implement a logger middleware that prints out the status code that was responded to the http client, but cannot access from the `httpd_req_t` struct nor from any function provided in `esp_http_server.h`

### Describe the solution you'd like.

signature functions like `httpd_resp_get_*` to get info from the response internal object, or make a public response object like `httpd_resp_t` with all the info like status code, the buffer that is going to be send to the client and its size_t, the headers that are going to be sended, etc.

Also as a plus, this is going to create a whole new API for sending http responses.

hmalpani, 2024-03-18:
Hello @otaxhu 
Apologies for the delayed response.

I think you can achieve this using `esp_http_client_config_t::event_handler`. When an event occurs, this handler gets triggered and you can log the required data from this handler. As you want to log the status code, I think you can log it in the handler as:

```
esp_err_t _http_event_handler(esp_http_client_event_t *evt)
{
...
ESP_LOGI(TAG, "status code: %d", esp_http_client_get_status_code(evt->client));
...

Please let me know if this helps.

otaxhu, 2024-03-18:
Client side no, I'm talking about HTTP server, but the idea of creating a event is very good in this case.

hmalpani, 2024-03-20:
You can also use event_handler in esp_http_server. For reference: https://docs.espressif.com/projects/esp-idf/en/latest/esp32/api-reference/protocols/esp_http_server.html#event-handling

hmalpani, 2024-04-01:
I hope using event_handler in esp_http_server helped resolve the issue. Closing this issue. Please feel free to reopen.

closed 2024-04-01 by hmalpani