espressif/esp-idf #13202: when build blufi example "blufi_init.c:35:5 unknown type name esp_bluedroid_config_t"

kuailefeifei, 2024-02-18:
### Answers checklist.

- [X] I have read the documentation [ESP-IDF Programming Guide](https://docs.espressif.com/projects/esp-idf/en/latest/) and the issue is not addressed there.
- [X] I have updated my IDF branch (master or release) to the latest version and checked that the issue is present there.
- [X] I have searched the issue tracker for a similar issue and not found a similar issue.

### IDF version.

v5.3-dev-1922-g5454d37d49

### Operating System used.

Windows

### How did you build your project?

Command line with Make

### If you are using Windows, please specify command line type.

None

### What is the expected behavior?

1. esp_bluedroid_config_t should be known in blufi_init.c
2. blufi example should be compiled normally.

### What is the actual behavior?

During compiling blufi_init.c report error unknown type name esp_bluedroid_config_t

### Steps to reproduce.

1. Step: cd esp-idf\examples\bluetooth\blufi
2. Step: idf.py set-target ESP32
3. Step: idf.py build

### Build or installation Logs.

```plain
[1194/1204] Building C object esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj

FAILED: esp-idf/main/CMakeFiles/__idf_main.dir/blufi_init.c.obj 

ccache D:\Programs\Espressif\tools\xtensa-esp32-elf\esp-12.2.0_20230208\xtensa-esp32-elf\bin\xtensa-esp32-elf-gcc.exe -DESP_PLATFORM -DIDF_VER=\"v5.1.2-dirty\" -DMBEDTLS_CONFIG_FILE=\"mbedtls/esp_config.h\" -DSOC_MMU_PAGE_SIZE=CONFIG_MMU_PAGE_SIZE -DUNITY_INCLUDE_CONFIG_H -D_GNU_SOURCE -D_POSIX_READER_WRITER_LOCKS -IE:/Projects/personal/esp-idf/examples/bluetooth/blufi/build/config -IE:/Projects/personal/esp-idf/examples/bluetooth/blufi/main -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/newlib/platform_include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/FreeRTOS-Kernel/portable/xtensa/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/esp_additions/include/freertos -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/esp_additions/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/freertos/esp_additions/arch/xtensa/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_hw_support/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_hw_support/include/soc -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_hw_support/include/soc/esp32 -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_hw_support/port/esp32/. -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_hw_support/port/esp32/private_include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/heap/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/log/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/soc/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/soc/esp32 -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/soc/esp32/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/hal/esp32/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/hal/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/hal/platform_port/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/include/esp32 -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_rom/esp32 -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_common/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/soc -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_system/port/include/private -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/xtensa/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/xtensa/esp32/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/include/apps -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/include/apps/sntp -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/lwip/src/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/port/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/port/freertos/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/port/esp32xx/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/lwip/port/esp32xx/include/arch -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_ringbuf/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/efuse/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/efuse/esp32/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/deprecated -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/analog_comparator/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/dac/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/gpio/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/gptimer/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/i2c/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/i2s/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/ledc/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/mcpwm/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/parlio/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/pcnt/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/rmt/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/sdio_slave/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/sdmmc/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/sigma_delta/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/spi/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/temperature_sensor/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/touch_sensor/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/twai/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/uart/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/usb_serial_jtag/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/driver/touch_sensor/esp32/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_pm/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/port/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/mbedtls/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/mbedtls/library -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/esp_crt_bundle/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/mbedtls/3rdparty/everest/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/mbedtls/3rdparty/p256-m -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/mbedtls/mbedtls/3rdparty/p256-m/p256-m -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_app_format/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/bootloader_support/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/bootloader_support/bootloader_flash/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_partition/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/app_update/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_mm/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/spi_flash/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/pthread/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_timer/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/app_trace/include -ID:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/esp_event/include -ID:/Programs/Espressif/f [... cut ...] ks/esp-idf-v5.1.2/components/newlib D:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/partition_table D:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/soc D:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/spi_flash D:/Programs/Espressif/frameworks/esp-idf-v5.1.2/components/xtensa
-- Configuring done
-- Generating done
-- Build files have been written to: E:/Projects/personal/esp-idf/examples/bluetooth/blufi/build/bootloader
ninja: build stopped: subcommand failed.
```

zhp0406, 2024-02-20:
@kuailefeifei 

I followed your method and was not able to reproduce the issue; the example compiled successfully. Here is my log:

```
zhanghaipeng@FA002426:~/esp/esp-idf/examples/bluetooth/blufi$ idf.py build
Executing action: all (aliases: build)
Running ninja in directory /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build
Executing "ninja all"...
[12/1249] Generating ../../partition_table/partition-table.bin
Partition table binary generated. Contents:
*******************************************************************************
# ESP-IDF Partition Table
# Name, Type, SubType, Offset, Size, Flags
nvs,data,nvs,0x9000,24K,
phy_init,data,phy,0xf000,4K,
factory,app,factory,0x10000,1500K,
*******************************************************************************
[501/1249] Performing configure step for 'bootloader'
-- Found Git: /usr/bin/git (found version "2.25.1") 
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/asio does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/cbor does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/coap does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/expat does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/libsodium does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/nghttp does not contain a CMakeLists.txt file. No component will be added
-- Component directory /home/zhanghaipeng/esp/esp-idf/components/tinyusb does not contain a CMakeLists.txt file. No component will be added
-- The C compiler identification is GNU 13.2.0
-- The CXX compiler identification is GNU 13.2.0
-- The ASM compiler identification is GNU
-- Found assembler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc
-- Check for working C compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-gcc -- works
-- Detecting C compiler ABI info
-- Detecting C compiler ABI info - done
-- Detecting C compile features
-- Detecting C compile features - done
-- Check for working CXX compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++
-- Check for working CXX compiler: /home/zhanghaipeng/.espressif/tools/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin/xtensa-esp32-elf-g++ -- works
-- Detecting CXX compiler ABI info
-- Detecting CXX compiler ABI info - done
-- Detecting CXX compile features
-- Detecting CXX compile features - done
-- Building ESP-IDF components for target esp32
-- Project sdkconfig file /home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/sdkconfig
-- Compiler supported targets: xtensa-esp-elf
-- Looking for sys/types.h
-- Looking for sys/types.h - found
-- Looking for stdint.h
-- Looking for stdint.h - found
-- Looking for stddef.h
-- Looking for stddef.h - found
-- Check size of time_t
-- Check size of time_t - done
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/soc/esp32/ld/esp32.peripherals.ld
-- Bootloader project name: "bootloader" version: 1
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.api.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.libgcc.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/esp_rom/esp32/ld/esp32.rom.newlib-funcs.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.ld
-- Adding linker script /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main/ld/esp32/bootloader.rom.ld
-- Components: bootloader bootloader_support efuse esp_app_format esp_bootloader_format esp_common esp_hw_support esp_rom esp_system esptool_py freertos hal log main micro-ecc newlib partition_table soc spi_flash xtensa
-- Component paths: /home/zhanghaipeng/esp/esp-idf/components/bootloader /home/zhanghaipeng/esp/esp-idf/components/bootloader_support /home/zhanghaipeng/esp/esp-idf/components/efuse /home/zhanghaipeng/esp/esp-idf/components/esp_app_format /home/zhanghaipeng/esp/esp-idf/components/esp_bootloader_format /home/zhanghaipeng/esp/esp-idf/components/esp_common /home/zhanghaipeng/esp/esp-idf/components/esp_hw_support /home/zhanghaipeng/esp/esp-idf/components/esp_rom /home/zhanghaipeng/esp/esp-idf/components/esp_system /home/zhanghaipeng/esp/esp-idf/components/esptool_py /home/zhanghaipeng/esp/esp-idf/components/freertos /home/zhanghaipeng/esp/esp-idf/components/hal /home/zhanghaipeng/esp/esp-idf/components/log /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/main /home/zhanghaipeng/esp/esp-idf/components/bootloader/subproject/components/micro-ecc /home/zhanghaipeng/esp/esp-idf/components/newlib /home/zhanghaipeng/esp/esp-idf/components/partition_table /home/zhanghaipeng/esp/esp-idf/components/soc /home/zhanghaipeng/es [... cut ...]
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash --flash_mode dio --flash_size 2MB --flash_freq 40m 0x1000 build/bootloader/bootloader.bin 0x8000 build/partition_table/partition-table.bin 0x10000 build/blufi_demo.bin
or from the "/home/zhanghaipeng/esp/esp-idf/examples/bluetooth/blufi/build" directory
 python -m esptool --chip esp32 -b 460800 --before default_reset --after hard_reset write_flash "@flash_args"
zhanghaipeng@FA002426:~/esp/esp-idf/examples/bluetooth/blufi$ git describe 
v5.3-dev-1922-g5454d37d49
```

zhp0406, 2024-02-20:
Is your submodule not up to date with the latest changes? Please update it using git submodule, then compile again.

zhp0406, 2024-02-28:
Has your issue been resolved?

closed 2024-03-10 by Alvin1Zhang